jacopotediosi/OctoPrint-Telegram #272: {time_finish} / ETA variable not working

Quintin9801, 2020-07-16:
Octoprint version: 1.4.0 Win10
Telegram Plugin version: 1.5.0
Telegram App Version: 6.2.1

When receiving zchange notifications, My {time_finish} variable does not show anything, even when close to the end of a print (after it as time to calculate ETA). Are there any fixes for this issue that I am missing? Is eta calculated from an external plugin?

[image]
[image]

ZChange Script:
```
{emo:<info>} Printing, *{percent}% done* 
*Extruder* {e1_temp}/{e1_target}.
Time Done: {time_done}
Time Left: {time_left}
Time Finish: {time_finish}

[OctoPrint](192.168.1.60:5000)
[OctoAnywhere](https://www.getanywhere.io/app/devices#)

/status
```

giloser, 2021-02-09:
hi is this still a problem?

Quintin9801, 2021-02-09:
@giloser No this is not a problem anymore, this was patched in a previous update. I forgot this thread was still open.

closed 2021-02-09 by Quintin9801