jacopotediosi/OctoPrint-Telegram #275: Emojis not displaying properly from 1.5.1

DodgeDeBoulet, 2020-07-31:
There seems to be an issue with including emojis in notifications. Rather than the emoji image, I'm getting something that isn't being translated into an emoji:

```
\U0001f419 \U0001F4A4 Shutting down. Goodbye.
```

I've turned them off for now.

giloser, 2020-08-07:
Is the option to send the message as combine or separate? if separate in how is configured the markup?
is which version of octoprint? and which version of the plugin?

thanks

DodgeDeBoulet, 2020-08-07:
Message is sent as combined. I have never used separate.

The version of the plugin is in the subject ... but for clarity's sake it is 1.5.1 installed from https://github.com/szobov/OctoPrint-Telegram/archive/master.zip. The issue has been present since 1.5.1 was made available to resolve the incompatibility with OctoPrint 1.4.1rc1 and newer. I'm currently on 1.4.2 final, released yesterday.

electricJP, 2020-08-11:
> Message is sent as combined. I have never used separate.
>
> The version of the plugin is in the subject ... but for clarity's sake it is 1.5.1 installed from [...]

I have the same issue. I use combined messages: 
`\U0001f680 Hello. I'm online and ready to receive your commands.`

DodgeDeBoulet, 2020-08-12:
I have discovered the issue and fixed it in my local installation. Starting at line 457 in ~/oprint/lib/python2.7/site-packages/octoprint_telegram/\_\_init\_\_.py, **self.emojis** is defined as:

```
		self.emojis = {
			'octo': 	'\U0001F419', #octopus
			'mistake': 	'\U0001F616',
			'notify': '\U0001F514',
			'shutdown' : '\U0001F4A4',
			'shutup':	 '\U0001F64A',
			'noNotify': '\U0001F515',
			'notallowed': '\U0001F62C',
			'rocket': '\U0001F680',
			'save': '\U0001F4BE',
			'heart': '\U00002764',
			'info': '\U00002139',
			'settings': '\U0001F4DD',
			'clock': '\U000023F0',
			'height': '\U00002B06',
			'question': '\U00002753',
			'warning': '\U000026A0',
			'enter': '\U0000270F',
			'upload': '\U0001F4E5',
			'check': '\U00002705',
			'lamp': '\U0001F4A1',
			'movie': '\U0001F3AC',
			'finish': '\U0001F3C1',
			'cam': '\U0001F3A6',
			'hooray': '\U0001F389',
			'error': '\U000026D4',
			'play': '\U000025B6',
			'stop': '\U000025FC'
		}
```
However, it appears to need to be defined as this:

```
		self.emojis = {
			'octo': 	u'\U0001F419', #octopus
			'mistake': 	u'\U0001F616',
			'notify': u'\U0001F514',
			'shutdown' : u'\U0001F4A4',
			'shutup':	 u'\U0001F64A',
			'noNotify': u'\U0001F515',
			'notallowed': u'\U0001F62C',
			'rocket': u'\U0001F680',
			'save': u'\U0001F4BE',
			'heart': u'\U00002764',
			'info': u'\U00002139',
			'settings': u'\U0001F4DD',
			'clock': u'\U000023F0',
			'height': u'\U00002B06',
			'question': u'\U00002753',
			'warning': u'\U000026A0',
			'enter': u'\U0000270F',
			'upload': u'\U0001F4E5',
			'check': u'\U00002705',
			'lamp': u'\U0001F4A1',
			'movie': u'\U0001F3AC',
			'finish': u'\U0001F3C1',
			'cam': u'\U0001F3A6',
			'hooray': u'\U0001F389',
			'error': u'\U000026D4',
			'play': u'\U000025B6',
			'stop': u'\U000025FC'
		}
```
I made the modification, ran python -m py_compile \_\_init\_\_.py (not sure this was necessary) and the issue is resolved for me.

rlogiacco, 2020-08-15:
Would you be willing to create a pull request for this?

DodgeDeBoulet, 2020-08-15:
> Would you be willing to create a pull request for this?

I would, but I believe the issue is already fixed in the source. From what I can see a release candidate hasn't been created, though.

giloser, 2020-10-12:
fix with 1.6.0

closed 2020-10-12 by giloser