jacopotediosi/OctoPrint-Telegram #276: Plugin Loading fails because of ImportError

Ecuashungo, 2020-08-04:
The plugin install was successful via the plugin manager. But upon reboot it does not load in the plugin manager and the log shows following error message:

> 2020-08-04 18:35:53,562 - octoprint.plugin.core - ERROR - Error loading plugin telegram
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 972, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 5, in <module>
    from flask.ext.babel import gettext
ImportError: No module named ext.babel

I'm using OctoPrint Version Version 1.4.1 and octopi version 0.17.0 to install everything. This is a fresh install.

cp2004, 2020-08-04:
See the release notes for OctoPrint 1.4.1.

This has been fixed in the latest version of the plugin, 1.5.1. You may have to 'force check for updates' or manually reinstall.

See also https://github.com/fabianonline/OctoPrint-Telegram/issues/269

Ecuashungo, 2020-08-05:
Thanks I followed instructions in #277, which worked.

closed 2020-08-05 by Ecuashungo