jacopotediosi/OctoPrint-Telegram #277: Not installing through the plugin manager.

grossqx, 2020-08-04:


cp2004, 2020-08-04:
See the release notes for OctoPrint 1.4.1.

This has been fixed in the latest version of the plugin, 1.5.1. You may have to 'force check for updates' or manually reinstall.

I believe this is because the listing in the plugin repo is targeting the branch `stable`, which has not been updated.

@giloser Please either merge the branch `master` to `stable` or update the plugin repo listing, ASAP

rogeliodh, 2020-08-04:
update fails with:

```
Updating, please wait.
/home/rogelio/OctoPrint/bin/python2 -m pip --disable-pip-version-check install https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip
ERROR: HTTP error 404 while getting https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip
ERROR: Could not install requirement https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip because of error 404 Client Error: Not Found for url: https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip
ERROR: Could not install requirement https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip for URL https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip
```

tobyfu, 2020-08-04:
I managed to get it updated by doing an install by URL with this link https://github.com/fabianonline/OctoPrint-Telegram/archive/1.5.1.zip

falsechicken, 2020-08-05:
> I managed to get it updated by doing an install by URL with this link https://github.com/fabianonline/OctoPrint-Telegram/archive/1.5.1.zip

Yep this works as a temporary workaround.

grossqx, 2020-08-05:
I can confirm that it works.

cp2004, 2020-08-05:
This issue should now be solved, as plugin repository has been updated.

 To @giloser or whoever is maintaining this plugin, **the plugin repo has been changed to target `master`, no longer `stable`** to work around this issue as it is very widespread, and causes issues for OctoPrint as tickets are opened there, complaining about this problem. 
If this needs changing please open an issue there.

closed 2020-10-12 by giloser