jacopotediosi/OctoPrint-Telegram #278: Pictures ignore horizontal and vertical flip

MettyD, 2020-08-05:
In Webcam-Settings of Octoprint the Options are checked, horizontal flip and vertical flip.
The images that get send with telegram are not flipped.

Gord-the-Rogue, 2020-08-06:
I'm having this same issue after updating to the latest version of Octoprint 1.4.1 and 1.4.2.
I'm hopeful the developers sees this and pushes an update for telegram to work with the latest version of octoprint.

giloser, 2020-08-07:
Hi did you Flip the camera directly in Octoprint webcam options? 
if yes you have to do the same configuration again in the Telegram plugin.
We just ask to Octoprint to send us the image and after that we rotate if configured.
The image is showed as octoprint send it and with the configuration you have done in the plugin.

I 'm still at 1.4.0 so I don't know if this has change in the most recent release.

MettyD, 2020-08-07:
Okay, but where do I have to make these settings in telegram plugin ? I cannot see any option for that in the settings.

giloser, 2020-08-07:
Okay I check again and we should take back the config from octoprint.

I'll try again when I get

giloser, 2020-08-07:
Sorry my bad you are right we use the global settings.
I have seen that someone event propose a fix for that I'll add this for the next release that I hope will be quicker that the last one.

MettyD, 2020-08-07:
Whish I could Phyton so that I could help you .. only php, Perl and others i am used to

cad435, 2020-08-07:
Well, I actually notice this strange behavior:

enabling "horizontal flip" flips the picture horizontally in telegram bot.
enabling "vertical flip" ALSO flips picture horizontally in telegram bot.

I use Octoprint-Telegram 1.5.1 and can't seem to find any "flip" button in it's config

cad435

cutnicace, 2020-08-24:
Hey there, just to throw in my two cents... 
Thanks @Sekisback for pointing out the code part for the quick fix.

In my case it was already negated and did not do what it should ;) so I had to remove the **not**
~~~if flipH or flipV or rotate:
                        image = Image.open(StringIO.StringIO(data))
                        #if not flipH: -> previous
                        if flipH:
                                image = image.transpose(Image.FLIP_LEFT_RIGHT)
                        #if not flipV: -> previous
                        if flipV:
                                image = image.transpose(Image.FLIP_TOP_BOTTOM)~~~

MettyD, 2020-08-24:
> as a workaround, on your raspberry open the file init.py located under /oprint/lib/python2.7/site-packages/octoprint_telegram
> id did it with sudo nano -c init.py scoll down to line 1402
>
> if flipH or flipV or rotate: image = Image.open(StringIO.StringIO(data)) if flipH: image = image.transpose(Image.FLIP_LEFT_RIGHT) if flipV: image = image.transpose(Image.FLIP_TOP_BOTTOM) if rotate: image = image.transpose(Image.ROTATE_270) output = StringIO.StringIO() image.save(output, format="JPEG") data = output.getvalue() output.close() return data
>
> change the if flipH and/or the if flipV to if not flipH or if not flipV
> save this with ctrl + s and exit with ctrl + x
> reboot the pi

Just for my understanding:

In these Lines of Code the Flags flipH, filpV and rotate get initialized:
flipH = self._settings.global_get(["webcam", "flipH"])
flipV = self._settings.global_get(["webcam", "flipV"])
rotate= self._settings.global_get(["webcam", "rotate90"])

They may contain true or false;

In the original coding it is asked if one of these is true.

if flipH or flipV or rotate:
                        image = Image.open(StringIO.StringIO(data))
                        if not flipH:
                                image = image.transpose(Image.FLIP_LEFT_RIGHT)
                        if not flipV:
                                image = image.transpose(Image.FLIP_TOP_BOTTOM)
                        if rotate:
                                image = image.transpose(Image.ROTATE_270)
                        output = StringIO.StringIO()
                        image.save(output, format="JPEG")
                        data = output.getvalue()
                        output.close()

What kind if sense does it make, that I do a flip or a rotate only, if its NOT flipped ? Rotate should work.

It should be like this, right ?

if flipH or flipV or rotate:
                        image = Image.open(StringIO.StringIO(data))
                        #if not flipH:
                        if flipH:
                                image = image.transpose(Image.FLIP_LEFT_RIGHT)
                        #if not flipV:
                        if flipV:
                                image = image.transpose(Image.FLIP_TOP_BOTTOM)
                        if rotate:
                                image = image.transpose(Image.ROTATE_270)
                        output = StringIO.StringIO()
                        image.save(output, format="JPEG")
                        data = output.getvalue()
                        output.close()

Best regards

JeefX, 2020-08-25:
Adding this here because I am seeing a similar issue. Pictures are working fine for me with both horizontal and vertical flip enabled, but gifs do not get flipped as they should. Images look fine for me, gifs are still upside down

MettyD, 2020-08-26:
> Adding this here because I am seeing a similar issue. Pictures are working fine for me with both horizontal and vertical flip enabled, but gifs do not get flipped as they should. Images look fine for me, gifs are still upside down

I fixed this issue quick an dirty: connect to pi via ssh.
cd /oprint/lib/python2.7/site-packages/octoprint_telegram
sudo vi __init__.py
type ESC and then type :1500 to go to line 1500.
Scroll down until you find this code:
params.append('cpulimit')
                        params.append( '-l')
                        params.append(  '65')
                        params.append(  '-f')
                        params.append(  '-z')
                        params.append(  '--')
                        params.append( 'ffmpeg')
                        params.append(  '-y' )
                        params.append( '-threads')
                        params.append( '1')
                        params.append(  '-i')
                        params.append( stream_url)
                        params.append(  '-t')
                        params.append(  timeSec)
                        params.append( '-c:v')
                        params.append( 'mpeg4')
                        params.append(  '-c:a' )
                        params.append( 'mpeg4')

at the end append the following lines:
                       # Rotation 180 dregree
                        params.append( '-vf')
                        params.append( 'transpose=2,transpose=2' )
                        # End Rotation

Save and restart the system. This worked for me, but is quick an dirty and does not check the CheckBoxes for Flip.

Best regards

JeefX, 2020-08-27:
That worked! It just flips everything so now my images are upside down and gifs are correct, but this is a good workaround for now. Thanks!

rlogiacco, 2020-08-27:
As I said, we have changed this code so many times it’s ridiculous. I would not consider this or any other snippet like that as a fix of any sort: we need to get to root or just leave it as broken...

MettyD, 2020-08-28:
It is just a workaround and no fix :) I am sure the developer will find a solution. Maybe foosel may help 😊

giloser, 2021-03-26:
please could you check if you can now make things work as expected on the latest version #352

f4nu, 2021-05-09:
@giloser the webcam (photos) seem to be OK now, the GIFs are still rotated tho.

TheNitek, 2022-01-06:
> 2022-01-06 15:05:12,763 - octoprint.plugins.telegram.TCMD - INFO - Will try to create a gif
> 2022-01-06 15:05:13,006 - octoprint.plugins.telegram - ERROR - Caught an exception trying clean previous images : [Errno 2] No such file or directory: '/home/pi/.octoprint/data/telegram/tmpgif/gif.mp4'
> 2022-01-06 15:05:13,007 - octoprint.plugins.telegram - INFO - test if nice exist
> 2022-01-06 15:05:13,007 - octoprint.plugins.telegram - INFO - test exist program 'nice', '--version'
> 2022-01-06 15:05:13,034 - octoprint.plugins.telegram - INFO - ret = 0
> 2022-01-06 15:05:13,035 - octoprint.plugins.telegram - INFO - test if cpulimit exist
> 2022-01-06 15:05:13,035 - octoprint.plugins.telegram - INFO - test exist program 'cpulimit', '--help'
> 2022-01-06 15:05:13,064 - octoprint.plugins.telegram - INFO - ret = 1
> 2022-01-06 15:05:13,066 - octoprint.plugins.telegram - INFO - test if ffmpeg exist
> 2022-01-06 15:05:13,066 - octoprint.plugins.telegram - INFO - test exist program 'ffmpeg', '-h'
> 2022-01-06 15:05:14,178 - octoprint.plugins.telegram - INFO - ret = 0
> 2022-01-06 15:05:14,430 - octoprint.plugins.telegram - INFO - sec=5
> 2022-01-06 15:05:14,431 - octoprint.plugins.telegram - INFO - timeSec=11:11:05
> 2022-01-06 15:05:14,432 - octoprint.plugins.telegram - INFO - limit_cpu=130.0 | used_cpu=2.0 | because nb_cpu=4
> 2022-01-06 15:05:14,432 - octoprint.plugins.telegram - INFO - Image transformations [H:False, V:False, R:True]
> 2022-01-06 15:05:14,432 - octoprint.plugins.telegram - INFO - Need to rotate 90deg counter clockwise
> 2022-01-06 15:05:14,433 - octoprint.plugins.telegram - INFO - will now create the video  'nice', '-n', '20', 'cpulimit', '-l', '130.0', '-f', '-z', '--', 'ffmpeg', '-y', '-threads', '2.0', '-i', 'http://localhost/webcam/?action=stream', '-t', '11:11:05', '-pix_fmt', 'yuv420p', '-vf', 'transpose=1', '/home/pi/.octoprint/data/telegram/tmpgif/gif.mp4'
> 2022-01-06 15:05:31,175 - octoprint.plugins.telegram - INFO - Finish the video
> 

Does this help?

miostreams, 2024-04-09:
This still does not appear to work. I have tried the fixes suggested throughout the comments, but the telegram images still appear to ignore octoprint flip settings in all cases.

Qualith, 2024-05-09:
Same here: I have my webcam installed facedown. 
In classic webcam plugin I have checked: Flip webcam horizontally and Flip webcam vertically and the live stream looks good inside octoprint and octoapp, but these options don't affect the image sent via Telegram that continues being facedown.

Gin154, 2025-04-12:
same here, I did the same as @Qualith and I got the same issue.