jacopotediosi/OctoPrint-Telegram #279: Telegram 1.5.1 still not working with 1.4.1 ?

silversegarra, 2020-08-05:
Hello, I updated OCtoprint to 1.4.1 and trying to install Telegram 1.5.1 plugin, everything seems to going well, but once rebooted, plugin isn't installed...
There's any solution?

Thanks

GpMissi, 2020-08-05:
Same problem

barman26, 2020-08-05:
same too!

silversegarra, 2020-08-05:
If we install manually, it works
https://github.com/fabianonline/OctoPrint-Telegram/issues/277

GpMissi, 2020-08-05:
> If we install manually, it works
> #277

No

silversegarra, 2020-08-05:
For me is working :/

SirFlip, 2020-08-05:
For me too. Thanks for the hint.
Hopefully it gets sorted out with the manager

GpMissi, 2020-08-05:
pi@octopi:~ $ pip install "https://github.com/fabianonline/OctoPrint-Telegram/archive/master.zip"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/fabianonline/OctoPrint-Telegram/archive/master.zip
  Downloading https://github.com/fabianonline/OctoPrint-Telegram/archive/master.zip
     - 501kB 6.2MB/s
    Complete output from command python setup.py egg_info:
    Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 255 in /tmp/pip-req-build-E39chw/

silversegarra, 2020-08-05:
Frome here
[image]

With this link
https://github.com/fabianonline/OctoPrint-Telegram/archive/1.5.1.zip

cp2004, 2020-08-05:
> pi@octopi:~ $ pip install "https://github.com/fabianonline/OctoPrint-Telegram/archive/master.zip"
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Collecting [...]

@GpMissi As the output says, you need to activate OctoPrint's python environment. Like `source ~/oprint/bin/activate`, then install the plugin. Or just use the plugin manager. 

For the benefit of anyone else here, the plugin manager has been updated to target `master` now, so hopefully it will begin to work again straight away.

PetrGasparik, 2020-08-05:
The trouble is, **Release tag 1.5.1 is missing release.zip** - there are just sources. That's all

It can be fixed manually, but update script for 1.4.1 depends on this. 

@giloser please fir Release tag with release.zip, thank you

cp2004, 2020-08-05:
@PetrGasparik You are indeed correct, this is because of the plugin's definition here https://github.com/fabianonline/OctoPrint-Telegram/blob/d8ef9da4f210ba43989a0845b0114c3340014ff8/octoprint_telegram/__init__.py#L922-L933

uses an explicit `release.zip` instead of the default `/archive/{target_version}.zip`, which is auto generated by Github. Endless issues!

Gord-the-Rogue, 2020-08-06:
I was able to reinstall it, but now my images are flipped in telegram and I can't figure out how to fix them. They show correctly in octoprint, but not telegram notifications. Anyone have this issue?

adamoutler, 2020-08-09:
Octoprint thinks the file is here
https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip
but it is actually here
https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.5.1/release.zip
note the swapped 1.5.1 and download

giloser, 2020-10-12:
fix with 1.6.0

closed 2020-10-12 by giloser