jacopotediosi/OctoPrint-Telegram #281: How to put gif to work (ffmpeg missing)

tiagomaricate, 2020-08-08:
Basically I'm receiving this message:

_Problem creating gif, please check log file, and make sure you have installed ffmpeg with following command : `sudo apt-get install ffmpeg_

Could someone help me?

giloser, 2020-08-08:
what is your version of octoprint, octopi and telegram plugin?
the command has to be run on the raspberry  as command line.
The same for the log you can see in ~/.octoprint/logs/octoprint.log

thanks

tiagomaricate, 2020-08-08:
> what is your version of octoprint, octopi and telegram plugin?
> the command has to be run on the raspberry as command line.
> The same for the log you can see in ~/.octoprint/logs/octoprint.log
> 
> thanks

Octoprint Version 1.4.0-rc6 (HEAD -> master branch)
Telegram plugin 1.5.1
I couldn't found the octopi version in any place.
I'm using an ocprint version that came with Robo R2 printer.
Not sure if this helps.

giloser, 2020-08-15:
like I said in the post before you'll have to open a command line on the raspberry and launch the command line : sudo apt-get install ffmpeg

closed 2020-11-13 by giloser