jacopotediosi/OctoPrint-Telegram #285: Installation failed on OctoPrint 1.4.2 Update

bemar76, 2020-08-14:
There seems to be something wrong with the update package.
What to do now?

`Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simpleCollecting https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zipERROR: HTTP error 404 while getting https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zipERROR: Could not install requirement https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip because of error 404 Client Error: Not Found for url: https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zipERROR: Could not install requirement https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip for URL https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip`

gary325, 2020-08-14:
This also REMOVED the current instance of OctoPrint-Telegram.
```
2020-08-14 07:45:06,073 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2020-08-14 07:45:06,075 > Collecting https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip
2020-08-14 07:45:06,077 ! ERROR: HTTP error 404 while getting https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip
2020-08-14 07:45:06,078 ! ERROR: Could not install requirement https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip because of error 404 Client Error: Not Found for url: https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip
2020-08-14 07:45:06,079 ! ERROR: Could not install requirement https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip for URL https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip
```
I can download the ZIP file, Local installation will be attempted after current print is finished.

gary325, 2020-08-14:
I installed the latest update via a locally download ZIP file, Install was successful and Telegram is communicating and sending out updates.

gary325, 2020-08-16:
GIF no longer works without adding in cpulimit. The GIF creation in the prior version worked OK.

giloser, 2020-08-16:
yes and it will stay like that as without it could break the print and/or the print.
the cuplimit was added to be more secure with that.

thinkl33t, 2020-08-16:
Looks like the path its trying to install from is wrong:

```
https://github.com/fabianonline/OctoPrint-Telegram/releases/1.5.1/download/release.zip
```

I used the following to install the correct version on octopi:

```
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.5.1/release.zip --no-cache-dir
```

aschwoerer, 2020-09-17:
Same issue here, after the failed upgrade the plugin was uninstalled from perspective of Octoprint. After a restart of Octoprint I just reinstalled the plugin via UI. Installer found the already installed version, uninstalled it and installed the new one:

> 2020-09-17 16:20:29,373 > Successfully built OctoPrint-Telegram
> 2020-09-17 16:20:29,373 > Installing collected packages: OctoPrint-Telegram
> 2020-09-17 16:20:29,374 > Found existing installation: OctoPrint-Telegram 1.5.0
> 2020-09-17 16:20:29,374 > Uninstalling OctoPrint-Telegram-1.5.0:
> 2020-09-17 16:20:29,374 > Successfully uninstalled OctoPrint-Telegram-1.5.0
> 2020-09-17 16:20:29,375 > Successfully installed OctoPrint-Telegram-1.5.2

Everything is up and running again, all preferences were kept. :)
Seems like a problem with the source-url in the updater to me.

closed 2020-10-12 by giloser