jacopotediosi/OctoPrint-Telegram #289: problems with Octolapse plugin

NacheteVBGamer, 2020-09-26:
Hi thank you very much for your work, your plugin is really good and useful, i have been working with it a lot of time and is really cool.

But recently i have some problems since I installed the 0.4.0 version of the octolapse.

The telegram bot sends me a notificacion of the status every minut or 2 times in a minut.

after watching the octoprint working i can see the octolapse bar there is an icon of a camera, it is of white color, but when it change his color to green (i think is when the octolapse plugin take a snapshot) i receive inmediatly 2 new captures of status in my telegram bot.

I don`t know if is any problem of my configuration or if i can disable this function of receive in my telegram bot two captures when octolapse makes a snapshot.

Thank you very much for all

Deses, 2020-10-11:
Probably a duplicate of https://github.com/fabianonline/OctoPrint-Telegram/issues/13

NacheteVBGamer, 2020-10-12:
> 
> 
> Probably a duplicate of #13 or similar root issue.

i don´t know what it is, how can i solution   it?

lethuer, 2022-01-14:
I have the exact same problem with telegram notifications !

lethuer, 2022-04-25:
> I have the exact same problem with telegram notifications !

Again I enabled Octolapse plugin and remembered this bug with many telegram notifications. Is there any progress in solving this issue ?