jacopotediosi/OctoPrint-Telegram #290: Time and Height notification is not working

ViiFii, 2020-10-03:
Hello.
Plugin installed and respond to /status etc commands.. but configured notifications to registered users are not working at all... Didn't find any solution so, post request here.
[image]

TobiasBales, 2020-10-07:
same for me

chatainsim, 2020-11-12:
Same here too!

ViiFii, 2021-01-02:
Solved

closed 2021-01-02 by ViiFii

TobiasBales, 2021-01-24:
@ViiFii so how was it solved? Would be good to know really because for me everything is up-to-date but it still does not work

ViiFii, 2021-01-24:
> @ViiFii so how was it solved? Would be good to know really because for me everything is up-to-date but it still does not work

hi. just press "balloon" under "Notify" and set all you need ;-)