jacopotediosi/OctoPrint-Telegram #292: Notification after cooldown

n3roGit, 2020-10-04:
It would be very nice if we had the option to get informed if the bed temperature is at a specific level after print.
Something like:
Print is finished and cooled down.

Deses, 2020-10-11:
Duplicate of https://github.com/fabianonline/OctoPrint-Telegram/issues/286.

giloser, 2021-02-22:
it is possible now with the plugin [https://github.com/gmccauley/OctoPrint-CooldownNotification](url)  that will send a gcode when a temp is reached if you put something like `M118 E1 UserNotif Bed should be at 35°`  in that plugin you should get a User Notif notification

n3roGit, 2021-02-25:
Unfortunately, this does not work for me. Should I be able to send directly to Telegram via the GCODE

giloser, 2021-02-25:
did you activate the new notification UserNotif ?

n3roGit, 2021-03-16:
How does it work?

giloser, 2021-03-17:
In the plugin option for your user you have to activate the notification like Zchanged, printStart or printDone 
you have now in the list one 'UserNotif' that you need to activate to receive the message
and then in the plugin  https://github.com/gmccauley/OctoPrint-CooldownNotification you can define a gcode to be sent when the printer reach the wanted temperature and for that gcode you put something like
**M118 E1 UserNotif** Bed should be at 35° 
the bold part is the needed part and the rest is up to you.

hope this is more clear

n3roGit, 2021-03-31:
[image]
Ahh okay its a little bit hidden.
but after enabeling it, it doesnt work :(

maybe becouse of my klipper firmware?

giloser, 2021-04-01:
I would not say it's hidden it's like that that all the notifications works and you had to enable it when you configure it.

have you installed the plugin I talk about earlier? configured correctly? so you saw the gcode being send to your printer with the config I talk about?

I don't know if klipper can be a probleme for this.