jacopotediosi/OctoPrint-Telegram #293: incompatible message 1.5.2

TGM3D, 2020-10-05:
I used Telegram very often up to Octoprint version 1.4.1. I installed the new Image Octopi Version 0.17.0 with Octoprint 1.4.2. The plugin manager can no longer find Telegram. Installed via URL, it always says: Incompatible with Octoprint or Phyton. However, query Phyton results in 2.7.16 This is less than 3. and greater than 2.7 as specified. Where is the mistake? How can you help yourself?
The whole thing on several Octopi installations is the same ...

TGM3D, 2020-10-05:
I found the solution here.  https://github.com/fabianonline/OctoPrint-Telegram/pull/291

Deses, 2020-10-11:
> I found the solution here. #291

If you have the solution then please close this issue. 👍

giloser, 2020-10-11:
this should be fixed with the new 1.6.0 version

closed 2020-10-11 by giloser