jacopotediosi/OctoPrint-Telegram #298: Error while updating to latest 1.6.0beta1 (from working 1.5.2)

spanzetta, 2020-10-11:
These are the relevant log outputs 
any idea on why it happens?
Thanks

**octoprint.log**
2020-10-11 18:19:30,075 - octoprint.plugins.softwareupdate - INFO - Starting update of telegram to 1.6.0...
2020-10-11 18:19:32,470 - octoprint.plugins.softwareupdate - ERROR - Update of telegram can not be performed, please also check plugin_softwareupdate_console.log for possible causes of this
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1155, in _perform_update
    update_result = updater.perform_update(target, populated_check, target_version, log_cb=self._log, online=online)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py", line 116, in perform_update
    raise exceptions.UpdateError("Error while executing pip install", (stdout, stderr))
UpdateError
2020-10-11 18:19:32,710 - octoprint.plugins.tracking - INFO - Sent tracking event update_failed, payload: {'to_version': '1.6.0', 'from_version': '1.5.2', 'target': u'telegram'}
2020-10-11 18:20:03,180 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:79.20.129.224

**plugin_softwareupdate_console.log**
2020-10-11 18:19:30,085   /home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.6.0/release.zip --no-cache-dir
2020-10-11 18:19:32,461 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2020-10-11 18:19:32,463 > Collecting https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.6.0/release.zip
2020-10-11 18:19:32,467 ! ERROR: HTTP error 404 while getting https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.6.0/release.zip
2020-10-11 18:19:32,468 ! ERROR: Could not install requirement https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.6.0/release.zip because of error 404 Client Error: Not Found for url: https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.6.0/release.zip
2020-10-11 18:19:32,469 ! ERROR: Could not install requirement https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.6.0/release.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.6.0/release.zip for URL https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.6.0/release.zip

Deses, 2020-10-11:
It happens because there's no release 1.6.0 zip in github.

Either they forgot to build the release files or there was something wrong and they pulled the files. Maybe they want you to manually build it for testing... I dunno lol

giloser, 2020-10-11:
It should be fixed now , the release.zip was indeed missing.

closed 2020-10-11 by giloser

spanzetta, 2020-10-11:
I confirm it works!
Thanks