jacopotediosi/OctoPrint-Telegram #303: 1.6.0 beta1- No notifications

jmarc67, 2020-10-13:
Octopi 0.17.0 / Octoprint 1.4.2
Python 2.7.16

Hello,

I upgraded from 1.5.2 ( working without issue) to 1.6.0 beta1 and i had errors and no nore telegram notifications.

After a successful 1.6.0 installation in my octoprint. log i have:

- octoprint.server.api - ERROR - Error calling SimpleApiPlugin telegram
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 71, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 992, in on_api_get
    'bind_cmd':[k for k, v in self.tcmd.commandDict.items() if 'bind_none' not in v ],
AttributeError: 'NoneType' object has no attribute 'commandDict'
- tornado.access - ERROR - 500 GET /api/plugin/telegram?bindings=true (::ffff:10.0.0.202) 22.62ms

I came back to 1.5.2 : notifications are working

Is 1.6.0 beta 1 only working with python 3+ ?

Thanks for your help.

giloser, 2020-10-13:
Hi,

no the 1.6.0.beta.1 should work on both python 2.7 and higher

did you set the command to your bot?
should have something like that in Telegram
[image]

jmarc67, 2020-10-13:
No, nothing up with Telegram (1.6.0b)  No problem with 1.5.2

I found this in my log with 1.60b1:

octoprint.plugins.telegram - ERROR - Exception on_startup: 'NoneType' object has no attribute 'port'

Something wrong with port.

giloser, 2020-10-13:
can you put more of the log file?

I don't think the error on the port could bring the behavior you seems to have
when you update to 1.6.0 the token is set correctly and verified?

thanks

jmarc67, 2020-10-13:
Token was set correctly and verified with 1.6.0

I'll try 1.6.0 one more time and will give you log file.

jmarc67, 2020-10-13:
giloser, here is my .log with 1.6.0 :

[octoprint.log](https://github.com/fabianonline/OctoPrint-Telegram/files/5370670/octoprint.log)

(I do not have these errors with 1.5.2)

giloser, 2020-10-13:
sorry but can you try again with the checkbox from the bottom of the plugin configuration page checked

[image]

thanks

jmarc67, 2020-10-13:
.log just after checking debug information.
Reboot

[octoprint.log](https://github.com/fabianonline/OctoPrint-Telegram/files/5370782/octoprint.log)

giloser, 2020-10-13:
thanks a lot.

Can you try to desactivate the emoticon to see if the problem could come from them?

thanks

jmarc67, 2020-10-13:
I tried, nothing new. No notifications.

szobov, 2020-10-13:
@jmarc67 Hey mate,
May I ask you to run this : `python -m pip install https://github.com/szobov/OctoPrint-Telegram/archive/fix-utf-encoding-python2.zip` on the environment with octoprint?

jmarc67, 2020-10-13:
@szobov Hello ;)

Done but no notification.

szobov, 2020-10-13:
hm... super-strange.
Because it's rendered fine in your log:
```
2020-10-13 13:35:18,469 - octoprint.plugins.telegram - DEBUG - Formatting emoticon: `rocket`
2020-10-13 13:35:18,472 - octoprint.plugins.telegram.TMSG - DEBUG - Sending Notification: 🚀 Hello. I'm online PI 3 and ready to receive your commands.
2020-10-13 13:35:18,477 - octoprint.plugins.telegram - DEBUG - Send_msg() found event: PrinterStart
2020-10-13 13:35:18,480 - octoprint.plugins.telegram - INFO - Sending a message: 🚀 Hello. I'm online PI 3 and ready to receive your commands. with_image=True with_gif=0 chatID= 729362975
2020-10-13 13:35:18,482 - octoprint.plugins.telegram - DEBUG - data so far: {u'disable_notification': False, u'chat_id': '729362975', u'disable_web_page_preview': True}
```

jmarc67, 2020-10-13:
I dont know how to resolve that.

I have to go to work. I'll test after.

Thanks for your help

szobov, 2020-10-13:
yes, thank you for understanding.
I'll try to realize what can we do to fix it.

jmarc67, 2020-10-14:
Nothing new this morning.
Yesterday evening i had to print an item and reverted one more time to 1.5.2. All was working fine.
I put the new 1.6.0b if i can test something. I don't know why i have no notification with only the last 1.6.0 

Something to add: my Linux octopi is 4.19.118-v7+ #1311 SMP Mon Apr 27 2020 should i go to the last 5.4.51 ?

If you have an idea

giloser, 2020-10-14:
Hi, I'm sorry but I have no idea right now why it was working before and not now.

so you are on a linux PC and not on a raspberry right?

Like it's said in the release the 1.6.0 is a beta, it's a first version of the migration to python3 and still working with python2.

jmarc67, 2020-10-14:
No i'm on a Raspberry Pi3 B.

Ok. No problem giloser. I'll stay with 1.5.1 and later move to python 3.x with 

curl -L https://get.octoprint.org/py3/upgrade.py --output upgrade.py
python3 upgrade.py

Thanks for all.

szobov, 2020-10-14:
> No i'm on a Raspberry Pi3 B.
> 
> Ok. No problem giloser. I'll stay with 1.5.1 and later move to python 3.x with
> 
> curl -L https://get.octoprint.org/py3/upgrade.py --output upgrade.py
> python3 upgrade.py
> 
> Thanks for all.

But I still want to find what is the problem :)
I'm sorry I wan't able to prepare debugging version, but if you would have some time to test maybe on this week, it would be super-cool.

jmarc67, 2020-10-14:
@szobov Yes It would be nice if this version could work on my PI.

Test is life :)

jmarc67, 2020-10-15:
@szobov Here are some news !

What i did:

Upgraded to the last kernel (from Linux octopi 4.19.118-v7+ #1311 SMP Mon Apr 27 2020 to Linux octopi 5.4.51-v7+ #1333 SMP Mon Aug 10 2020) -> Nothing new. No Notifications.

Upgraded from Python 2.7.16 to Python 3.7.3 (with script here https://octoprint.org/blog/2020/09/10/upgrade-to-py3/ )

All is OK now !

So i had telegram notifications problem with python 2.x and now it's gone with 3.x

I don't know because it was not working with old python for me.

closed 2020-10-15 by jmarc67

szobov, 2020-10-15:
It's amazing, but we still have a problem for python2 users unfortunately :\
But I'm glad you fixed the problem.
Cheera