jacopotediosi/OctoPrint-Telegram #305: 1.6.0 emojis won't work

thiagolages, 2020-10-13:
Hi, 

I've seen this problem happen before but the reported solutions
- Leave NO space between emojis and text;
- Leave TWO spaces between emojis and text;
are not working on this version. I also tried leaving 3 and 4 spaces
but nothing seems to work.
I'd like to know if there are any other solutions or at least
a list of emojis that are not working at the moment.

Greatly appreciated.

giloser, 2020-10-14:
Hi,
could you be a little bit more specific? 

none of the emoticon works?

which version of octopi? of octoprint ? of python?

can you provide a log file with the debug option activate ?

thanks

thiagolages, 2020-10-14:
"None of the emoticons work" means whenever I get a message from my printer, it shows something like this:
"ERROR: I was not able to format the Notification for '' properly. Please open yout OctoPrint settings for Telegram Notifications and check message settings for 'TYPE' ", where TYPE can be either 'ZChange', 'PrinterStart', 'PrinterShutdown', etc.

- OctoPi 0.16.0
- OctoPrint 1.4.2 
- Python 2.7.13 

I ended up removing all emoticons by unchecking the "Send Emojis" box and
now I get the messages I configured but, of course, without emojis.
[octoprint.log](https://github.com/fabianonline/OctoPrint-Telegram/files/5380041/octoprint.log)

szobov, 2020-10-15:
Hi @thiagolages 
Can you please run: `python -m pip install https://github.com/szobov/OctoPrint-Telegram/archive/fix-utf-encoding-python2.zip` on the environment with octoprint?
It seems like the problem is in system encoding, because I see in your logs just bytes instead of emoji icons.
Cheers

szobov, 2020-10-15:
@thiagolages if it's possible, could you please contact me on telegram or in the plugin's telegram chat.
It would be much more easier to communicate there for quick debug.
Thanks

ulipo, 2020-10-17:
Same thing here, it worked util last update.
I tried to installe the fix-utf-encoding package to no avail.

thiagolages, 2020-10-17:
> Hi @thiagolages
> Can you please run: `python -m pip install https://github.com/szobov/OctoPrint-Telegram/archive/fix-utf-encoding-python2.zip` on the environment with octoprint?
> It seems like the problem is in system encoding, because I see in your logs just bytes instead of emoji icons.
> Cheers

I've executed the command under `/home/pi/oprint/bin` , but still no luck.

> @thiagolages if it's possible, could you please contact me on telegram or in the plugin's telegram chat.
> It would be much more easier to communicate there for quick debug.
> Thanks

Do you mean the "Octoprint-Telegram-Users"? I've joind the channel, maybe we can do this quick communications there, thanks for letting me know.
Also, does upgrading it to python3 solve the problem? Let me know if it does, I'd be glad to upgrade, thought it was already running under python3.

szobov, 2020-10-19:
> Also, does upgrading it to python3 solve the problem? Let me know if it does, I'd be glad to upgrade, thought it was already running under python3.

Afaik, yes it works. There are no open issues with python3.

> Do you mean the "Octoprint-Telegram-Users"? I've joind the channel, maybe we can do this quick communications there, thanks for letting me know.

Yes, let's try to solve the issue there. I'll DM you.

szobov, 2020-10-19:
> I tried to install the fix-utf-encoding package to no avail.

@ulipo how did you tried to install it?

ulipo, 2020-10-19:
Same way as Thiago, using ~/oprint/bin/pip
It installed but nothing changed, even after reboot

thiagolages, 2020-10-19:
> > Do you mean the "Octoprint-Telegram-Users"? I've joind the channel, maybe we can do this quick communications there, thanks for letting me know.
> 
> Yes, let's try to solve the issue there. I'll DM you.

Ok, Iet's talk on the group.
Thanks for the availability.

draik, 2020-10-19:
I'm having the same issue with Telegram messages containing mardown emotes, and it's no different after running the pip installation.
I've tried both:

`~/oprint/bin/python -m pip install https://github.com/szobov/OctoPrint-Telegram/archive/fix-utf-encoding-python2.zip`
and
`~/oprint/bin/pip install https://github.com/szobov/OctoPrint-Telegram/archive/fix-utf-encoding-python2.zip`

The output is the same:

```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/szobov/OctoPrint-Telegram/archive/fix-utf-encoding-python2.zip
  Using cached https://github.com/szobov/OctoPrint-Telegram/archive/fix-utf-encoding-python2.zip

["Requirement already satisfied" output block]

Building wheels for collected packages: OctoPrint-Telegram
  Building wheel for OctoPrint-Telegram (setup.py) ... done
  Created wheel for OctoPrint-Telegram: filename=OctoPrint_Telegram-1.6.0-py2-none-any.whl size=95341 sha256=1b317fcd06b4c8198bc0236f3092613da8910c3c5f6313a4015bdba04a7fcdb3
  Stored in directory: /home/pi/.cache/pip/wheels/5c/44/47/a5f0e12237a52490374681eca8d890a1eb29f78495d9419171
Successfully built OctoPrint-Telegram
```

draik, 2020-10-19:
This is the debug log entries from octoprint.plugins.telegram in octoprint.log

```
2020-10-18 14:12:33,099 - octoprint.plugins.telegram - DEBUG - Formatting emoticon: `finish`
2020-10-18 14:12:33,100 - octoprint.plugins.telegram.TMSG - DEBUG - Exception on formatting message: 'ascii' codec can't encode character u'\U0001f3c1' in position 0: ordinal not in range(128)
2020-10-18 14:12:33,100 - octoprint.plugins.telegram.TMSG - DEBUG - Sending Notification: ⚠ ERROR: I was not able to format the Notification for 'PrintDone' properly. Please open your OctoPrint settings for Telegram Notifications and check message settings for 'PrintDone'.
2020-10-18 14:12:33,101 - octoprint.plugins.telegram - DEBUG - Send_msg() found event: PrintDone
2020-10-18 14:12:33,102 - octoprint.plugins.telegram - INFO - Sending a message: ⚠ ERROR: I was not able to format the Notification for 'PrintDone' properly. Please open your OctoPrint settings for Telegram Notifications and check message settings for 'PrintDone'. with_image=False with_gif=0 chatID= [CHAT_ID]
2020-10-18 14:12:33,103 - octoprint.plugins.telegram - DEBUG - data so far: {u'disable_notification': False, u'parse_mode': 'Markdown', u'chat_id': '[CHAT_ID]', u'disable_web_page_preview': True}
2020-10-18 14:12:33,103 - octoprint.plugins.telegram - DEBUG - Sending without image.. [CHAT_ID]
2020-10-18 14:12:33,872 - octoprint.plugins.telegram - DEBUG - Sending finished. 200
```

ulipo, 2020-10-19:
I upgraded Octoprint to run on Python3 (via the script at https://github.com/cp2004/Octoprint-Upgrade-To-Py3 ) and now emotes are working again.

draik, 2020-10-19:
@ulipo thank you for that. I just did the same, and emotes are successfully working again.

draik, 2020-10-19:
> I upgraded Octoprint to run on Python3 (via the script at https://github.com/cp2004/Octoprint-Upgrade-To-Py3 ) and now emotes are working again.

@ulipo thank you for that. I just did the same, and emotes are successfully working again for me, too.

giloser, 2020-10-20:
Hi there,

I would like to try a different way to see if this could fix the problem
Could you please try to install the version from this link
https://github.com/giloser/OctoPrint-Telegram/archive/Fix_Emo_Python2.zip
You can install with octorprint in the plugin manager there is an option to install directly from link
[image]

giloser, 2020-10-20:
no sorry wait I have detect a problem on it I'll send a new message when it's ready.

sorry

giloser, 2020-10-21:
Hi there,

I would like to try a different way to see if this could fix the problem
Could you please try to install the version from this link
https://github.com/giloser/OctoPrint-Telegram/archive/Fix_Emo_Python2.zip
You can install with octorprint in the plugin manager there is an option to install directly from link

I hope this will help.

Jon-So, 2020-10-23:
I am having this same exact problem except now the images are coming out blurry too...

szobov, 2020-10-23:
> I am having this same exact problem except now the images are coming out blurry too...

Hi @Jon-So 
Did you try this branch @giloser suggested above?
I mean it's OK to write that you have the same issue, but why didn't you try to help to debug this issue? :)

Jon-So, 2020-10-23:
I have to wait for my print to finish so I can do this... just got about a few hours to go...

thiagolages, 2020-10-23:
> Hi there,
>
> I would like to try a different way to see if this could fix the problem
> Could you please try to install the version from this link [...]

I saw your previous message and was about to test when you told me to wait.
Now I've got another print going and will (again) wait to check if your solution works.
But I guess after trying it (even if it works), I'll probably switch to Python3. 
Do you guys think it's a good ideia or not yet? Maybe it can break other plugins, idk

giloser, 2020-10-23:
I think most of the plug-in have been adapted yet but you have one plug-in that is called 'python 3 version check' or something similar that will verify if everything is ready.

giloser, 2020-11-04:
Hey there,

any luck about the test of this version?

thanks

Jon-So, 2020-11-04:
Sorry I have been sick and not at my printers for the last few days, I will let you know as soon as I do... thanks for messaging me.

giloser, 2020-11-09:
hey no problem take care of yourself it's more important!

giloser, 2020-11-13:
This should be fix with the version 1.6.1 and 1.6.2.
if there is still problem we will open the ticket again.

closed 2020-11-13 by giloser

thiagolages, 2020-11-13:
Hi, I'm so sorry for not being active on the issue, but this has worked for me.
I did as you suggested, pasted the URL inside the package manager, and it started to work again.

Thanks for the fix !

> Hi there,
>
> I would like to try a different way to see if this could fix the problem
> Could you please try to install the version from this link [...]