jacopotediosi/OctoPrint-Telegram #307: Shutdown message not working

spanzetta, 2020-10-17:
All the others are working.. but the shutdown one doesn't work (it was working in the past.. then stopped)
If I reboot my Raspberry PI3A+ I get the shutdown message.. but not with shutdown 
Anyone else has this problem?
Which type of log can be useful to understand the cause of this issue?
Thanks in advance

ulipo, 2020-10-17:
Looks like last update broke the emoji system, if you remove {emo:octopus} and similar from the message, it will work.

spanzetta, 2020-10-18:
Thanks for the reply. 

The strange thing is that it doesn't happen always (I did some more tests)

Any chance it will be fixed by the author?

giloser, 2020-10-19:
Could you activate the debug and then send us the log when the problem has appears again?

thanks

Jon-So, 2020-10-23:
Mine does it too and it say to do something like Audi apt-get install cpulimit and when I do that nothing changes...

giloser, 2020-10-23:
ok could please activate the debug and send the log file when the problem occurs?
this problem has nothing to do with cpulimit , cpulimit is only needed to get gif working and not taking down the PI.

If the problem is fixed when you remove the emoji then it's probably related to this bug #305 and then please could you try what suggest in the topic and let we know if it help? and if not send the log file again.

thanks

giloser, 2021-02-09:
is this still a problem? or could we close it?

thanks

closed 2021-02-18 by giloser

spanzetta, 2021-07-02:
Hi, again I am experiencing the same issue..
Any idea why it happens?

This is the current configuration details:

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:89.0) Gecko/20100101 Firefox/89.0
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1400
env.hardware.ram: 385024000
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 3 Model A Plus Rev 1.0
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 19.3.1
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.1
printer.firmware: Marlin Creality 3D
systeminfo.generator: systemapi