jacopotediosi/OctoPrint-Telegram #308: GIF returns a blurry picture only.

MaikelKelder, 2020-10-23:
CPU limit installed and ffmpeg, latest versions.

Everything works, only /gif and /supergif dont, the take a long time and return a blurred picture.
/status works fine, sharp picture.

No idea what to do to solve this....
Octoprint 1.4.2
Telegram 1.6.0

any pointers in the right direction are appreciated!

giloser, 2020-10-23:
Hi,

could you please activate the debug checkbox and send a log file when you have tried to create a gif? 
Also maybe add the problematic gif and an correct image so we can see the problem?

thanks

MaikelKelder, 2020-10-23:
[octoprint (1).log](https://github.com/fabianonline/OctoPrint-Telegram/files/5429487/octoprint.1.log)

It saves the GIF as "GIF.mp4" is that correct? cant upload it here, but noticed it when saving it as a file on my phone.
Renamed it to be able to upload it here to GIF.PNG.GIF
On my PC this plays and looks ok. in telegram its a still blurr picture, added screenshot.
[image]
[image]
[image]

giloser, 2020-10-23:
hi, it doesn't work on the iphone right?
already heard that the gif don't work nicely on iphone but I have no idea why and no iphone to test that behavior ...

I'll try to think about it and found something

MaikelKelder, 2020-10-23:
Yup, its a iphone thing!! a friend used my bot and shows GIF just fine. Closing this one...
I will leave it open to post my fix if I find it soon..... for others who suffer the same issue.

Thanks for your help Giloser!!

MaikelKelder, 2020-10-23:
Friend of mine runs an older version of Telegram but also other things are older (octoprint, python)
Asking a gif from his printer works fine.... so its a combination of iphone/well... something on the PI..

giloser, 2020-10-29:
Hi,

I made a change and try to test with friend with Iphone. It seems to work for them let me know if it work also for you.
You can install a test version with this link
https://github.com/giloser/OctoPrint-Telegram/archive/Fix_blurry_gif_iphone.zip
just when you are in octoprint -> options-> plugin manager -> add new -> you should have the possibility to install from an URL,
you put the https://github.com/giloser/OctoPrint-Telegram/archive/Fix_blurry_gif_iphone.zip in and install.

thanks

MaikelKelder, 2020-10-31:
Checked.... and works!!! nice job!!
Took longer for me to test it, had a very long print going on...
Really appreciate it!

closed 2020-11-13 by giloser