jacopotediosi/OctoPrint-Telegram #315: Not working gif in Android

Melancholic, 2020-11-12:
Hello! After update to v1.6.1 GIFs stopped working in Android (i checked on some different devices), but this GIFs work great on MacOS (i don't have a IPhone for check).

Maybe, it was broken to #308

Screenshot from Android: 
[image]

Screenshot from MacOS: 
[image]

giloser, 2020-11-12:
Hi,

thats strange because I work with several android device and I didn't get any problem with the gif :(

can you tell me a little bit more about the android you use? version and model? version of telegram apps?

thanks

Melancholic, 2020-11-12:
Sure!

Device#1
`OnePlus 6, Android 10, Telegram v7.2.1`

Device#2
`Xiaomi A1, Android 9, Telegram v7.2.1`

OctoPrint version : `1.4.2`
OctoPi version    : `0.17.0`
ffmpeg version `4.1.4-1+rpt1~deb10u1`
CPUlimit version `2.4`

I got this troubles right after update Telegram notification plugins at 10.11.20 (I've last gif from 10.11.20 at 01.30a.m UTC+3)

Melancholic, 2020-11-12:
II just downgraded now to version `1.6.0` and GIFs again worked. It's good for me, but it means that last release had broken. If i'll have a free time - i'll try fix it (unless somebody not does do it early)

giloser, 2020-11-12:
I'll take a look tomorrow but I find that really strange that I don't have this problem.
The solution will probably to have an option to switch to the creation of gif like the version 1.6.0 or 1.6.1

do you think it's possible you send me one of the wrong gif on telegram?

thanks

giloser, 2020-11-13:
Hi!
ok so I still haven't been able to reproduce it on my android device but I made a change in the creation.

Could you try the following version ?
https://github.com/giloser/OctoPrint-Telegram/archive/Test_Gif.zip

@MaikelKelder could you try it as well?

thanks

Melancholic, 2020-11-13:
@giloser Hi, sorry for waiting!
How i can find you in telegram for sending broken gif?

Melancholic, 2020-11-13:
@giloser This fix not working for me(. 
And i getting "status message" instead supergif now (i getting message with current print state by command `supergif` =])

giloser, 2020-11-13:
hi this should link you to me

https://t.me/giloser

closed 2020-11-13 by giloser