jacopotediosi/OctoPrint-Telegram #318: Update to 1.6.2 not succesful (suggest URL fix)

H3ndrik-Jan, 2020-11-13:
Hi,

Trying to update from version OctoPrint-Telegram 1.5.0 left me with following error message:

`ERROR: HTTP error 404 while getting https://github.com/fabianonline/OctoPrint-Telegram/releases/1.6.2/download/release.zip
ERROR: Could not install requirement https://github.com/fabianonline/OctoPrint-Telegram/releases/1.6.2/download/release.zip because of error 404 Client Error: Not Found for url: https://github.com/fabianonline/OctoPrint-Telegram/releases/1.6.2/download/release.zip
ERROR: Could not install requirement https://github.com/fabianonline/OctoPrint-Telegram/releases/1.6.2/download/release.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/fabianonline/OctoPrint-Telegram/releases/1.6.2/download/release.zip for URL https://github.com/fabianonline/OctoPrint-Telegram/releases/1.6.2/download/release.zip`

I think this might be caused by an incorrect URL. Instead of "https://github.com/fabianonline/OctoPrint-Telegram/releases/1.6.2/download/release.zip", I think it should be "https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.6.2/release.zip".
This is also the URL that the github web interface leads to when downloading OctoPrint-Telegram v1.6.2.

Thanks in advance!

supercazzola, 2020-11-18:
Agree, there is a URL issue, and the above recommended URL from @H3ndrik-Jan seems to be correct

giloser, 2020-11-19:
Hi sorry I didn't get any problem with that until now and I don't know where or how I should change this link.

supercazzola, 2020-11-19:
In my particular setup, on a Robo3d R2 unit, the fact that octoprint detects there is an update available happens. When it tries to execute the update, it points back to the URL as F3bian points out. After the PIP fails due to not finding the file it is looking for, you can manually correct the URL at the command line and it will install.
For me, the install executes (and removes the old version, installs the new version), but the new version doesn’t place a shortcut under settings - installed plugins. So I opened a new issue. Thanks 

-----

The problem with the world is that the intelligent people are full of doubts, while the stupid ones are full of confidence.

giloser, 2020-11-19:
ok @supercazzola thanks from your other message I get you had octorprint 1.4.0 RC6 what is your version @H3ndrik-Jan ??

Could this be more related to the octoprint version? or pip version? 

I did the version as I did the previous one and like @supercazzola I just get the notification and click update and everything works out but I am on octoprint 1.4.2.

supercazzola, 2020-11-19:
I should add that I also and stuck at the version of octoprint, believe. Like with The marlin firmware,  Robo3d made so many changes to the default branch and then abandoned the printer and also the C2. I have no way to safely update the  software. I’m not surprised the plug-in doesn’t act as one would expect since the version of octorprint is stuck in the past. 

-----
The problem with the world is that the intelligent people are full of doubts, while the stupid ones are full of confidence.

H3ndrik-Jan, 2020-11-20:
@giloser I was using octoprint 1.4.0 as well.
Furthermore:
octopi version: 0.17.0
System: Pi 3B+
Pip: 19.3.1
Python 2.7.16

supercazzola, 2020-11-21:
for me:
pip 20.2.4 from /home/pi/oprint/local/lib/python2.7/site-packages/pip (python 2.7)
Octoprint Version 1.4.0-rc6 (HEAD -> master branch)

jonk999, 2020-11-22:
Hit the same issue when upgrading Octoprint from version 1.4.0
Restarted Octoprint and it appeared as though the plugin was no longer installed. I assume it uninstalled as part of the upgrade. Went into the plugin store and reinstalled and all appears okay. Still had the settings, etc.

H3ndrik-Jan, 2020-11-22:
I also did what @jonk999 said, with the same result. I updated octoprint from version 1.4.0 to 1.4.2, restarted the system after which the telegram plugin disappeared. I could then also install the latest version using the plugin manager and everything, including settings, worked again on octoprint 1.4.2.

giloser, 2021-02-09:
was there still a problem with the update of 1.6.3?

closed 2021-02-18 by giloser