jacopotediosi/OctoPrint-Telegram #319: Notifications Significantly Delayed and High Ram usage with MultiCam

sushibagel, 2020-11-15:
Hello, Thanks for the great plugin. 
I did, however, notice when using multiple cameras my notifications are very delayed and I get 75% + ram usage on my 4gb pi 4. I realize there is extra computing power needed to do multiple images but it seems pretty excessive.

giloser, 2020-11-15:
Hi
Can you tell us which version of the plug-in and of octoprint?
You do images or gif? Because there is warning about that for gif.
Thanks

sushibagel, 2020-11-15:
No gifs just images, Octoprint is 1.4.2 plugin is 1.6.2

giloser, 2020-11-16:
ok and you have seen that it take more ram when you active multicam on the plugin or just add a second camera and multicam on octoprint?

sushibagel, 2020-11-16:
> ok and you have seen that it take more ram when you active multicam on the plugin or just add a second camera and multicam on octoprint?

The additional ram usage seems to only happen when I have Multicam activated on the telegram plugin. Doesn't seem to impact it too greatly if only the multicam plugin is on.

giloser, 2020-11-17:
and the usage of ram is always higher or only when he tried to get the images?

giloser, 2020-11-24:
so this issue seems to come from an URL that is not format as the usual and then as we tried to get an image we call the stream and we loop.
we will have to wait  until @fabianonline  and his change on the plugin multicam are validated so we can get the camera snapshot a safer way.

jacopotediosi, 2025-07-11:
I hope this issue is resolved with the new webcam handling introduced in version 1.7.0. Since a malformed URL is mentioned, I believe it was also a user error.