jacopotediosi/OctoPrint-Telegram #320: FR- Version without CPULIMIT for GIF creation.

MaikelKelder, 2020-11-16:
Curerntly cpulimit install is needed, and ffmpeg process is limited to 100 by default.
This is 100% on one core, and it uses that, yet, creating a gif takes almost 5 minutes now on a Pi 3, but also a Pi 4.

The older version could run on all cores 100% and creating a GIF took a minute.

Since cpulimit is (old as shit) very limited in functionality, I tested it but only runs in command temporary on 3 cores, and is way faster. But you cannot store this, since the PID is different every time it creates a gif.

Is it possible to get a version without the need of cpulimit, OR get it to run on 3 cores?
The resources taken to run a print is about 15% cpu in octoprint, so it can easily take up 3 cores for gif creation.

Hope something is possible, yes, cant seem to wait 5 minutes on a gif :-p 
Also, its less nice, it has big gaps in frames now, and used to be smooth as hell....

Thanks for the plugin and previous support (on gif iphone thing) its appreciated.

giloser, 2020-11-16:
Hi the problem is not everybody got a pi 4 and a lot of people blames the gif creation for some failed print.
So I tried to limit the usage of CPU to avoid that.
it was already the case in the test version I did for the Iphone problem did you experience such a problem in that version as well?

MaikelKelder, 2020-11-16:
No, had absolutely no issues whatsoever, and my printer runs 15 hours a day atleast...
No issues with the hotfix patch, or the updated version... 
I have much resources left, looking in TOP, dont see how it could cause an issue when you only take one core of four available.
Maybe if they run other plugins that take too much resources?

MaikelKelder, 2020-11-16:
A friend of mine still runs older vresions on a 3B+ without cpulimit, runs gif creation all the time, me too on his printer, and no issues there, even without cpulimit. a gif creation takes only 30sec on his. (yes very jealous ;-)

MaikelKelder, 2020-11-16:
PS, I am using a Pi 3b (not +) and I have at least 80% of resources left during printing.

giloser, 2020-11-16:
Well I could try to add one more option that you could unaccivate to not use cpu limit but when you are saying older version which One?
And when I asked about the issue I was asking about the delay did you get such a delay with the 1.6.0 or 1.16.1 or the test version I did?

I never really check the time it take for me but never 5min.... Now I run on a rpi 4 4gb so...

MaikelKelder, 2020-11-16:
I think this longer creation of a GIF started in version 1.6.x my friend is running 1.5.0 and has no delay. (within a minute)

MaikelKelder, 2020-11-16:
Clocked it in detail..Telegram 1.6.1 with cpulimit default (100) on Pi 3B
/GIF takes 3 minutes and few secs.
/SUPERGIF takes 5 minutes and 30 secs.

with CPUlimit on 400 
/gif under 1 minute
/supergif in about 2 minutes

Strange thing is, CPU never gets higher than 50%/54% but I do notice my LEDs turn off for a second sometimes without cpulimit, yet no pause, stutter or interruption on printing.

giloser, 2020-11-17:
can you send me the command you tried and the cpu usage you get because I thinks that's strange

with cpu limit to 260 I get peek to 150% in CPU so ... 

also I would like a log file of the creation in octoprint please to check the time of creation of the video and the time to send it.

thanks

MaikelKelder, 2020-11-17:
I used:
cpulimit -l 300 -e ffmpeg
in TOP it does show 140% something, its just in Octoprint resources monitor , it only shows 50% (sorry for that)

[image]

giloser, 2020-11-17:
Ok sadly it's more like mine.
Could you contact me on telegram so we can discuss this faster and maybe then get a log file to check the time the creation of the video seems to take and check more info there? @giloser should get you to me.

thanks

giloser, 2020-11-23:
Hi,

can you try this version to see if it goes faster or not? and send me log file if it's not enough?
https://github.com/giloser/OctoPrint-Telegram/archive/Test_Gif.zip

thanks

MaikelKelder, 2020-11-23:
Seems about 33% faster than before!

without CPULIMIT - 120 sec - CPU max 100% (one core)
with CPULIMIT on 300 - 80 sec - CPU max 138% 

OLD version octoprint/telegram (no cpulimit needed) - 30 sec.

MaikelKelder, 2020-11-23:
Will send a log too, just not much time at the moment... 
To be clear, what log do you want, to make sure ....

giloser, 2020-11-24:
wait wait wait with the version I did you check the CPU usage ? was it over 100% or did you take the parameter and try by hand?

I would like a log or a part of log with the creation of gif inside the plugin.

thanks

MaikelKelder, 2020-11-24:
I tried your version but still have cpulimit installed.
One without any changes, the other with cpulimit on 300.

So what I said above, cpu usage was max 100% of one core when I dont change anything.
If I can remove cpulimit with this version, I will test that too.

giloser, 2020-11-24:
yes there is still cpu limit I tried to change some setting like use several thread instead of just one so we still stay low on CPU.

let's be clear, I don't think I'll completely remove CPULIMIT. if we let ffmpeg use the CPU as he want we really could have problem with the print.
The only option I see if you really want it without CPULIMIT is with a specific option and a big warning that said if you uncheck this , it could break your print.

MaikelKelder, 2020-11-24:
I totally agree, I think CPUlimit is needed, after all, the print is all that matters, so I dont think its worth it to make a version without it.
Yet.... having 4 cores, SHOULD give you atleast 3 cores for GIF creation, since printing takes just 40% max of a single core.
But I have never been able to get CPULIMIT to such numbers. even when I put it on 400 (4 cores 100%) ffmpeg only uses 150% max.

But it seems that its still using just on thread by default.

MaikelKelder, 2020-11-24:
[octoprint 24-11.log](https://github.com/fabianonline/OctoPrint-Telegram/files/5589966/octoprint.24-11.log)
Hope this helps.

giloser, 2020-11-24:
yeah on the test version I get the number of core and divide it by 2.

I have made a version (with the same link as before) where instead of limiting CPULIMIT to 65 I will multiply if by the number of CPU I allow to use.

so if you have 4 core (see in the log like it's your case) I let ffmpeg to use 2 thread and  CPULIMIT to 130 we will see if it's good enough for you and not breaking print.

I have read some info about TOP command on linux and it seems the 100% is for one CPU so if you have 4 it should be able to go up to 400%, but still let try to be reasonable and not use to much.
Also htop seems to give more info I'll try to test with those new info.

MaikelKelder, 2020-11-24:
Ah.. HTOP is indeed better....
Ans gif creation is at one minute now, and still see resources left, with latest test version.
Seems like you are getting to the ideal situation here ;-)
It takes 2 cores, highest peek being 185% so thats looking ideal.

giloser, 2020-11-24:
Ok I propose I let you with this for some days and if everything is good I will create a PR and add it to the master.

MaikelKelder, 2020-11-24:
Agreed, I will test it with my prints running.

giloser, 2021-02-09:
I think this is enough days :)

I'll close this one

closed 2021-02-09 by giloser