jacopotediosi/OctoPrint-Telegram #322: upgrade from Telegram Notifications (1.5.0) to 1.6.2 not successful on Octoprint Version 1.4.0-rc6

supercazzola, 2020-11-18:
Even after correcting the URL as noted by@H3ndrik-Jan,  [https://github.com/fabianonline/OctoPrint-Telegram/issues/318](url), the install appears to finish successfully, but the plugin never appears on the left side of the Settings window of Octoprint.
Is there a reason 1.5.0 works and 1.6.2 doesn't with this older version of Octoprint ?

giloser, 2020-11-19:
Hi,

I didn't get any problem with the link and I'm cannot say for the older version of Octoprint.

maybe you can send a log of the failed installation?

thanks

supercazzola, 2020-11-19:
I will try to get some logs, but the install doesn’t appear to fail (there are no messages to indicate problem). It is just that after install, after reboot of system (or restart of octoprint), the plugin is not listed under the left side, so you can not make any adjustments to it. 
Is it clear?
I can try to duplicate and save the logs, but the machine is currently in a long print, and as you know, the plugin manager is disabled during printing.
Thank you 

-----

The problem with the world is that the intelligent people are full of doubts, while the stupid ones are full of confidence.

giloser, 2020-11-19:
no problem it can wait

supercazzola, 2020-11-21:
Here is the log entry from when it appears to successfully install (after manually correcting the URL as F3bian mentioned and running command via command line ):

`2020-11-18 12:51:03,180   /home/pi/oprint/bin/python -m pip install https://github.com/fabianonline/OctoPrint-Telegram/releases/release.zip
2020-11-18 12:51:07,691 > Collecting https://github.com/fabianonline/OctoPrint-Telegram/releases/release.zip
2020-11-18 12:51:08,195 ! HTTP error 404 while getting https://github.com/fabianonline/OctoPrint-Telegram/releases/release.zip
2020-11-18 12:51:08,196 ! Could not install requirement https://github.com/fabianonline/OctoPrint-Telegram/releases/release.zip because of error 404 Client Error: Not Found for url: https://github.com/fabianonline/OctoPrint-Telegram/releases/release.zip
2020-11-18 12:51:08,204 ! Could not install requirement https://github.com/fabianonline/OctoPrint-Telegram/releases/release.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/fabianonline/OctoPrint-Telegram/releases/release.zip for URL https://github.com/fabianonline/OctoPrint-Telegram/releases/release.zip
2020-11-18 12:51:08,708 ! You are using pip version 9.0.1, however version 20.2.4 is available.
2020-11-18 12:51:08,709 ! You should consider upgrading via the 'pip install --upgrade pip' command.
2020-11-18 12:51:52,640   /home/pi/oprint/bin/python -m pip install https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.6.2/release.zip
2020-11-18 12:51:57,017 > Collecting https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.6.2/release.zip
2020-11-18 12:51:58,021 > Downloading https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.6.2/release.zip (109kB)
2020-11-18 12:52:04,302 > Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,303 > Requirement already satisfied: pillow in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,304 > Requirement already satisfied: Flask-Login<0.3,>=0.2.2 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Login-0.2.11-py2.7.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,304 > Requirement already satisfied: Flask-Babel<0.10,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Babel-0.9-py2.7.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,305 > Requirement already satisfied: scandir<1.4,>=1.3 in /home/pi/oprint/lib/python2.7/site-packages/scandir-1.3-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,306 > Requirement already satisfied: Flask-Assets<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/Flask_Assets-0.10-py2.7.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,306 > Requirement already satisfied: feedparser<5.3,>=5.2.1 in /home/pi/oprint/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,307 > Requirement already satisfied: rsa<3.3,>=3.2 in /home/pi/oprint/lib/python2.7/site-packages/rsa-3.2.3-py2.7.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,308 > Requirement already satisfied: PyYAML<3.11,>=3.10 in /home/pi/oprint/lib/python2.7/site-packages/PyYAML-3.10-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,308 > Requirement already satisfied: flask<0.11,>=0.9 in /home/pi/oprint/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,311 > Requirement already satisfied: tornado==4.0.2 in /home/pi/oprint/lib/python2.7/site-packages/tornado-4.0.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,312 > Requirement already satisfied: sarge<0.2,>=0.1.4 in /home/pi/oprint/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,313 > Requirement already satisfied: pkginfo<1.3,>=1.2.1 in /home/pi/oprint/lib/python2.7/site-packages/pkginfo-1.2.1-py2.7.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,314 > Requirement already satisfied: netifaces<0.11,>=0.10 in /home/pi/oprint/lib/python2.7/site-packages/netifaces-0.10.6-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,315 > Requirement already satisfied: Jinja2<2.9,>=2.8 in /home/pi/oprint/lib/python2.7/site-packages/Jinja2-2.8.1-py2.7.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,315 > Requirement already satisfied: chainmap<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages/chainmap-1.0.2-py2.7.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,316 > Requirement already satisfied: sockjs-tornado<1.1,>=1.0.2 in /home/pi/oprint/lib/python2.7/site-packages/sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,317 > Requirement already satisfied: future<0.16,>=0.15 in /home/pi/oprint/lib/python2.7/site-packages/future-0.15.2-py2.7.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
2020-11-18 12:52:04,318 > Requirement already satisfied: psutil<3.3,>=3.2.1 in /home/pi/oprint/lib/python2.7/site-packages/psutil-3.2.2-py2.7-linux-armv7l.egg (from OctoPrint->OctoPrint-Telegram==1.6.2)
[... cut ...]
2020-11-18 12:52:04,358 > Installing collected packages: OctoPrint-Telegram
2020-11-18 12:52:04,359 > Running setup.py install for OctoPrint-Telegram: started
2020-11-18 12:52:07,532 > Running setup.py install for OctoPrint-Telegram: finished with status 'done'
2020-11-18 12:52:07,533 > Successfully installed OctoPrint-Telegram-1.6.2
2020-11-18 12:52:08,203 ! You are using pip version 9.0.1, however version 20.2.4 is available.
2020-11-18 12:52:08,204 ! You should consider upgrading via the 'pip install --upgrade pip' command.
`

supercazzola, 2020-11-21:
So it appears to install, however, it never appears on the left side under installed plugins, so that you could set the options.

kornflex, 2020-12-11:
Hello,

Same problem for me with octoprint 1.5.2  :/

giloser, 2021-02-09:
hi there is this still a problem with the new version?

thanks

supercazzola, 2021-02-09:
> hi there is this still a problem with the new version?
> 
> thanks

If you are asking me, I can try sometime this week. But unless things changed, I can’t see it installing

giloser, 2021-02-09:
I asked anybody with the same problem yes.
and it's a new version so yes some things have changed.

billedeclown01, 2021-05-24:
hello i have the same issue with octoprint 1.6.1 
the plugin is installed successfully but i cant see it. When the install is over i have a notification from octoprint "install:unknown" insted of "install:<name of the plugin>"

jacopotediosi, 2025-07-10:
This issue concerns very old versions of OctoPrint. Since the plugin installs correctly on recent OctoPrint versions and no useful logs for debugging were provided, I will proceed to close the issue, hoping no one is still experiencing this problem.

closed 2025-07-10 by jacopotediosi