jacopotediosi/OctoPrint-Telegram #323: Rotated image

alasso, 2020-11-19:
Hi, 

I have octopi configured to flip horizontally and vertically the webcam image and also rotated 90 degrees counter clockwise.
With this configuration the stream on octopi web it is Ok but in telegram the image appears rotated 180 degrees.

My installation:
Octopi Version 1.4.2
Octoprint Telegram 1.6.2

Any help will be appreciated
Adrian

[image]

[image]

giloser, 2020-11-19:
Hi!

you have configured the flip in octopi (with the script file) or in octoprint (with the webconfig)?

thanks

alasso, 2020-11-22:
Hi giloser,

I configured the flip with webconfig. 
Have I to configure it in another place?

Thanks!

[image]

giloser, 2020-11-23:
Hi,
no this should be good.
Could you send me a log so I can check what seems to be wrong?
thanks

alasso, 2020-11-23:
Hi,

I resolved in a dirty way.

I edited /home/pi/oprint/lib/python2.7/site-packages/octoprint_telegram/__init__.py

At the end of take_image function I replace:
```
image = image.transpose(Image.ROTATE_270)
by
image = image.transpose(Image.ROTATE_90)
```

Do you still need the OctoPrint log file? 

Thanks a lot for your help!
Adrian

giloser, 2020-11-24:
hi ,

glad you fix your issue but this will still be overwrite every time you update.

I still think it's not logical the option in octoprint says counter clockwise so 270 should be the correct option.

closed 2021-02-09 by giloser