jacopotediosi/OctoPrint-Telegram #324: [Feature] Send timelapse after print finish

CommanderRedYT, 2020-11-20:
Plugin would detect OctoLapse-Instance, and if turned on, would send the timelapse as mp4 after rendering is finished.

giloser, 2021-02-09:
should be on the next release

giloser, 2021-02-18:
#342

closed 2021-02-18 by giloser

cruzer619, 2021-03-03:
tried 3 different prints.. never get notified of "movie done" when print completes.  Is there any other setting needs be set besides the notification check mark?

J3n50m4t, 2021-03-03:
How large are your timelapse Videos? If its over 20 mb nothing will Happen (See #347)

cruzer619, 2021-03-03:
ahhh its a 23mb file.. yah just saw the thread... ok gotcha.. but thought the limit was a 50mb file but as long as its being looked at.. i'm good.. will try to make a smaller timelapse to test..

cruzer619, 2021-03-03:
ok just made a 2.4mb timelapse and I didn't get a notification.. I have something set wrong...