jacopotediosi/OctoPrint-Telegram #328: Telegram disapears in octoprint 1.5.2

kornflex, 2020-12-11:
Hello,

After latest upgrade of octprint, I don't have notifications.
Telegram isn't show parameters screen.

Thank you for your help

Mostum, 2020-12-19:
Hi,

i had the same issue.
Uninstalled the plugin via SSH:
`/home/pi/oprint/bin# ./pip uninstall Octoprint-Telegram`
installed it again and cleared its plugindata in the pluginmanager..

Had to enter the api key again, worked fine after.

timsweb, 2021-01-12:
I believe I'm having the same issue. I've installed 1.6.2 to Octopi 1.5.2, which appears to install successfully, but the plugin does not appear within the Octoprint UI.

giloser, 2021-01-12:
Hi! I don't understand why it could still be a problem at first for me it was the fact that you had python3 and the plugin didn't work with the 3 but now ....

Could you provide the plugin_manager log and the octoprint log?

thanks

Zwaubel, 2021-01-12:
I'm having the same issue. 
@Mostum's proposal does not work for me. The plugin is never listed as installed nor does the plugin manager appear to see any leftovers after uninstalling via `./pip uninstall Octoprint-Telegram`.

Here are my logs:
- plugin manager: [plugin_pluginmanager_console.log](https://github.com/fabianonline/OctoPrint-Telegram/files/5802805/plugin_pluginmanager_console.log)
- octoprint: [octoprint.log](https://github.com/fabianonline/OctoPrint-Telegram/files/5802802/octoprint.log)

giloser, 2021-01-12:
thanks, 

could you try to uninstall and reinstall once again please?

Zwaubel, 2021-01-12:
Unfrotunatly your commit didn't fix the current issue.
Here are my logs:
[plugin_PrintTimeGenius_engine.log](https://github.com/fabianonline/OctoPrint-Telegram/files/5802985/plugin_PrintTimeGenius_engine.log)
[octoprint.log](https://github.com/fabianonline/OctoPrint-Telegram/files/5802986/octoprint.log)

giloser, 2021-01-12:
sorry I put a space by correcting and didn't see it myself.
Hope this is finally ok now

Zwaubel, 2021-01-12:
Awesome, now it installed just fine. Thank you very much! 🎉

timsweb, 2021-01-12:
Working for me now too. 👍

kornflex, 2021-01-12:
> Hi,
> 
> i had the same issue.
> Uninstalled the plugin via SSH:
> `/home/pi/oprint/bin# ./pip uninstall Octoprint-Telegram`
> installed it again and cleared its plugindata in the pluginmanager..
> 
> Had to enter the api key again, worked fine after.

I have restore à backup and telegram comes :)

closed 2021-02-09 by giloser