jacopotediosi/OctoPrint-Telegram #329: Pre and post snapshot GCODE

sqweezr, 2020-12-12:
I am a total noob in programming......but i was wondering if it would be possible to have a pre and post snapshot gcode (like OctoLapse) so i can turn on a strip of LEDs for the snapshot as my printer is in a dark place.

Kind regards,

Koen

TeeVau, 2021-01-08:
I need something similar. But instead of a gcode I have to control a GPIO from the raspberry.

sqweezr, 2021-01-23:
I am also using it to control a GPIO on the raspberry pi. I am already using it in octolapse, when the plugin makes a picture it uses a pre snapshot gcode to turn on a strip of LEDs and a post snapshot gcode to turn them off again, so it is possible.

TeeVau, 2021-01-24:
How did you realize this? Which gcode can control a griot from raspberry? I thought that gcode is controlling the 3D printer.

sqweezr, 2021-01-24:
Man, I am sorry. I am indeed controlling a GPIO on my Ramps 1.4 board not on the raspberry pi

giloser, 2021-02-08:
check version 1.6.3

sqweezr, 2021-02-08:
Thanks a lot!!! This is exactly what i was looking for. Works like a charm

closed 2021-02-08 by giloser