jacopotediosi/OctoPrint-Telegram #332: [Issue]Frequent DNS queries

ljasinskipl, 2021-01-04:
Telegram plugin spams my dns with 2 queries per minute for api.telegram.com. I think it should be cached somewhere. It's by far the most frequent query on my dns

Pavulon87, 2021-02-18:
> Telegram plugin spams my dns with 2 queries per minute for api.telegram.com. I think it should be cached somewhere. It's by far the most frequent query on my dns

I know that feeling.
It's also my number one.

[image]

giloser, 2021-02-19:
you are welcome to propose something.
for me the call to the api.telegram.org is to check if the bot receive any message but I can be wrong

closed 2025-06-24 by jacopotediosi