jacopotediosi/OctoPrint-Telegram #334: Unable to install the plugin on Windows 10 machine

mburke1246, 2021-01-14:
I have been unable after using both the Plugin Manager and the console window to install this plugin. All indications from both paths is that it worked without issue, but the plugin does not show up after I restart Octoprint. Here is the command that I was running to install this manually

c:\OctoPrint\venv\Scripts>pip install https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.6.2/release.zip

I was able to get 1.5.2 to install and work at one point, but I would really like to get the latest one working if possible,

Thanks!

giloser, 2021-01-15:
Hi 

do you have any debug log from the octorpint.log and plugin_manager.log?

thanks

draggOP, 2021-02-03:
I have the same issue, i'm trying to install OctoPrint-Telegram via plugin manager, the installer says done and a prompt appears saying a restart is required for the following plugins -Installed: Unknown.
And after the restart octoprint telegram is not installed.
Here are my logs:
[octoprint.log](https://github.com/fabianonline/OctoPrint-Telegram/files/5917454/octoprint.log)
[plugin_pluginmanager_console.log](https://github.com/fabianonline/OctoPrint-Telegram/files/5917455/plugin_pluginmanager_console.log)

giloser, 2021-02-03:
thanks for the log.

I have seen one problem in the log and made a change in my test version.
Could you try to install this one? and tell me if it works?
https://github.com/giloser/OctoPrint-Telegram/archive/master.zip

thanks

draggOP, 2021-02-03:
Just installed, it's now working flawlessly!
Thanks

giloser, 2021-02-08:
thanks so this should be fixed with 1.6.3

closed 2021-02-08 by giloser