jacopotediosi/OctoPrint-Telegram #335: Can't be updated via Octoprint due to wrong link to release

FixxCZ, 2021-01-26:
Update return error:
ERROR: HTTP error 404 while getting https://github.com/fabianonline/OctoPrint-Telegram/releases/1.6.2/download/release.zip
becase this link is wrong.

Correct link it 
https://github.com/fabianonline/OctoPrint-Telegram/releases/download/1.6.2/release.zip

download and 1.6.2 are swapped in the link.

webaugur, 2021-01-26:
I was able to workaround this issue by manually downloading release.zip and upgrading via "install from zip" option in plugin manager. I upgraded from 1.5.0 to 1.6.2 using this workaround.

FixxCZ, 2021-01-27:
Ok, thank you, I will install it directly using zip file.

giloser, 2021-01-31:
Hi,

I have no idea where I should change this link. I never got error when updating.
What version of pip are you using? what version of octoprint?

webaugur, 2021-02-02:
This change broke upgrades from the previous version to this version (1.5.0 cannot upgrade to 1.6.2 for sure):
https://github.com/fabianonline/OctoPrint-Telegram/commit/e04b926fa47847b6d49cb2a7581f08cd42b13393

giloser, 2021-02-03:
great thanks ,

I'll try another link for the next update

giloser, 2021-02-09:
I hope this is fixed?

webaugur, 2021-02-11:
If you have 1.5.0 or older installed the update link is wrong in your installation. You will need to manually install 1.5.1 or newer in order to obtain the fixed code. Future updates will work after manually update.

The fix can be found in this commit prior to the 1.5.1 release: https://github.com/fabianonline/OctoPrint-Telegram/commit/e04b926fa47847b6d49cb2a7581f08cd42b13393

giloser, 2021-02-18:
I hope I finally help with this problem and we can close this.
open it or create a new one if the problem is still present.

closed 2021-02-18 by giloser