jacopotediosi/OctoPrint-Telegram #339: [Feature Request] message for a generic gcode

SM-26, 2021-02-11:
Hello!

this is a follow up to feature request we had on telegram user group.

TL;DR
new telegramMsgDict that is more generic and user controlled.

Long story:
I do a lot of sequential printing.
I would like to know when the printer is done with each object, but I don't need it to stop/pause in order to tell me that.
(So m600 would not work)
Just a telegram notification is all I need.

I've read the [Wiki](https://github.com/fabianonline/OctoPrint-Telegram/wiki/Add%20commands%20and%20notifications) but I must admit it's too complicated for me.

Looking at [Cancle Object Plugin](https://github.com/paukstelis/Octoprint-Cancelobject#prusaslicersuperslicer)
[image]
 this is all we need?

but, other idea I have, is maybe let the use select a g-code he would like to get notification about. i.e. M300 (beep)?

In addition, here are full list(s) of placeholders in prusaSlicer, if that's help [GithubWiki](https://github.com/prusa3d/PrusaSlicer/wiki/List-of-all-placeholders-taken-from-the-code) & [PrusaKB](https://help.prusa3d.com/en/article/macros_1775)

giloser, 2021-02-12:
Hi thanks a lot

like I said in the chat I think a good way could be to use the M118 gcode command as it used to send a message through the serial connection
me could do a `M118 E1 OBJECT name of the object`
that will be seen in the serial connection
`2021-02-12 10:03:37,239 - Recv:  T:15.63 /0.00 B:15.63 /0.00 @:0 B@:0
2021-02-12 10:03:38,789 - Send: M118 E1 OBJECT name of the object
2021-02-12 10:03:38,799 - Recv: echo:OBJECT name of the object
2021-02-12 10:03:38,800 - Recv: ok
2021-02-12 10:03:39,239 - Recv:  T:15.63 /0.00 B:15.63 /0.00 @:0 B@:0`
so we can can the `echo:OBJECT` as the type of message and `name of object` as the content
and after that we can imagine to add different type of `echo:XXX` for different type of message

SM-26, 2021-02-13:
I've tested ver 1.6.4:

I've added ```M118 E1 UserNotif objectName``` in between each object.
Attached gcode file. [DremelSeq_0.15mm_PLA_MINI_1h13m.gcode](https://github.com/fabianonline/OctoPrint-Telegram/files/5976161/DremelSeq_0.15mm_PLA_MINI_1h13m.gcode.txt)
While looking at the gcode file, it seems that the name of the object is specified after the skirt is completed.... this is kind of unfortunate, right?

I've rebooted my Pi, then after everything is booted, I've stared a sequential print. But it doesn't seem to work
[image]

log from Pi
[octoprint (1).log](https://github.com/fabianonline/OctoPrint-Telegram/files/5976182/octoprint.1.log)

SM-26, 2021-02-14:
[image]
note to self (and others) be sure to enable the new notification type!

Initial testing is looking good, but I want to do a bit more testing.

SM-26, 2021-02-15:
I've done more testings
in prusaSlicer, we have a placeholder for object number
``` M118 E1 UserNotif [complete_objects] done, let's do the next one ```
[image]

very nice indeed!

giloser, 2021-02-18:
#342

closed 2021-02-18 by giloser