jacopotediosi/OctoPrint-Telegram #340: Upload file fails with `Something went wrong during processing of your file`

candera, 2021-02-14:
OctoPrint version : 1.5.3
OctoPi version    : 0.18.0
Telegram plugin: 1.6.3

When attempting to upload a file via Telegram, the plugin responds with `Something went wrong during processing of your file`, regardless of file size.

Zipping the file, as suggested elsewhere, also fails.

I've attached an example of a gcode file that fails: [quill-lock.zip](https://github.com/fabianonline/OctoPrint-Telegram/files/5978083/quill-lock.zip). I've tried both that zipfile and the gcode file inside of it.

Octoprint log contains this error message:

```
2021-02-14 11:48:43,599 - octoprint.plugins.telegram.listener - ERROR - Exception occured during processing of a file: Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_telegram/__init__.py", line 273, in handleDocumentMessage
    f.write(data)
TypeError: write() argument must be str, not bytes
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_telegram/__init__.py", line 273, in handleDocumentMessage
    f.write(data)
TypeError: write() argument must be str, not bytes
```

Any assistance would be appreciated - I really like the Telegram plugin, and use it for most of my prints. Not being able to upload files is a bummer.

giloser, 2021-02-14:
hi

yes I think I have found the solution for this problem I 'll try to create a new version as soon as possible.

If you want to test the solution you can try this version 
https://github.com/giloser/OctoPrint-Telegram/archive/master.zip

and let me know if it fix the problem 😉

candera, 2021-02-14:
Upload seems to work now, thanks! I do see this in the logs still:

```
2021-02-14 17:30:03,857 - octoprint.plugins.telegram.listener - INFO - gcode
2021-02-14 17:30:45,075 - octoprint.filemanager.analysis - INFO - Starting analysis of local:TelegramPlugin/quill-lock-upper-20210214-1723.gcode
2021-02-14 17:30:45,314 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python3 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/TelegramPlugin/quill-lock-upper-20210214-1723.gcode
2021-02-14 17:30:49,428 - octoprint.plugins.telegram.listener - ERROR - Exception occured during remove of a file= /home/pi/.octoprint/data/telegram/tmpzip/quill-lock-upper-20210214-1723.gcode : Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_telegram/__init__.py", line 286, in handleDocumentMessage
    os.remove(path)
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/.octoprint/data/telegram/tmpzip/quill-lock-upper-20210214-1723.gcode'
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_telegram/__init__.py", line 286, in handleDocumentMessage
    os.remove(path)
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/.octoprint/data/telegram/tmpzip/quill-lock-upper-20210214-1723.gcode'
(END)
```

giloser, 2021-02-15:
well to be able to work with preprocessor of gcode like cancelobject or displaylayerprogress I had to change the way we upload the file.
I don't know why but upload the file as a stream doesn't work and we have to save the file locally before upload.
so I do save the file locally and try to remove it after to avoid space disk problem
seems that he doesn't found the file to delete it.

Could you please check if there is file on the path '/home/pi/.octoprint/data/telegram/tmpzip/' ?
If there is no file there is probably no problem ...

candera, 2021-02-15:
There are several files at that location, but not the one referenced in the log message.

giloser, 2021-02-18:
so I think we will close this one and if you still have the problem with the delete file you can create a new one.
don't forget to install the original version now that the fix is not that release.

closed 2021-02-18 by giloser