jacopotediosi/OctoPrint-Telegram #341: First time using - Refresh chat list not working 1.6.3

Canhobix, 2021-02-15:
After adding my Telegram Bot Token, and have my connection status: Connected as "@mybotname", I'm able to test the Token and it says that the Token Status is Valid for my "mybotname".
But when I reload the chat list, nothing happens and 
[image]

And in the Telegram **Fabians Octoprinter Bot** I got this message: **Now I know you. Before you can do anything, go to OctoPrint Settings and edit some rights.**

Thank you so much for your support.

giloser, 2021-02-15:
did you talk to your bot or the one from fabians? search for the @Barrxxx bot you are connected and try to talk to that one and then check again

Canhobix, 2021-02-15:
Hello, Yes I did, I talked from my bot to fabians one vice versa.
Plz correct me if I'm wrong, but I open my bot and tried /status and also @fabianonline_octoprinter_bot to issue a /status command, and I still get the same msg,  Before you can do anything, go to OctoPrint Settings and edit some rights.

Canhobix, 2021-02-15:
Never mind guys, my apologies, it was a browser issue, I'm using Brave and it seems that the browser tokens can't be read from the OctoPrint page because it changes to avoid being track, but I just browsed from MS Edge (chrome based) and worked.
Take care.

closed 2021-02-15 by Canhobix