jacopotediosi/OctoPrint-Telegram #344: html and markdown

gorobey, 2021-02-17:
html and markdown syntax not works, message will be sent as nosrmal text

giloser, 2021-02-17:
can you be more specific?
did you tried with an image or the message alone?
can you give me an example of what you tried?

gorobey, 2021-02-19:
@giloser, i have found a workaround:
the problem is that the option button for select messages forma have inline hidden attribute, but works, i have disabled with devoloper console, but on reload obviously elements return hidden.

giloser, 2021-02-20:
thanks but again what did you tried?
the syntax is forced when there is a image and the the message tried to send the image with the message in one message
so you did you tried with or without image?
combined or separate?

gorobey, 2021-06-25:
@giloser i have disabled images (not have a webcam connected to octoprint), this hide the buttons.
I have "solved" using browser console.

closed 2025-06-24 by jacopotediosi