jacopotediosi/OctoPrint-Telegram #346: Automatic Message Every 5 Minutes

GandalftheGay, 2021-02-22:
i really like this plugin so far! i just got one Problem:
i set the the "Send notification every" setting to 5mm and 5 minutes but it does not send me the messages at all.
i print over SD currently(started via Octoprint though) is that the problem?
Sure, it cant detect the layer it is at via SD-printing but it can still see the printtime and the plugin recognizes that its printing
/status works without a problem.

giloser, 2021-02-22:
yes the problem is that we send a message when we receive a message from the printer and the time is used to not send too many message there is no timer running to say now it's time to send a message.
and like describe in the octoprint documentation when running from SD card no message are send. At least not many and not the one we would needed to send a message.

giloser, 2021-03-26:
#352 can you check if the option in the latest version works for you?

jacopotediosi, 2025-07-11:
I consider this one solved by #352.