jacopotediosi/OctoPrint-Telegram #348: [Feature Request] Support for 12 Hour time formatting

Quintin9801, 2021-02-26:
I've been using this plugin for years and I do rely on it to update me on my printers' progress, and I love it. It would be nice to have a toggle between 24h time and 12h time to display an easier reading format for those who struggle switching between the two in their head (me). This would only apply to the `{time_finish}` variable in the status updates.

For example instead of the `{time_finish}` variable outputting "13:29:37", it would output "1:29:37 pm". I have tried switching the locale to display in 12h format, but this does not apply to the plugin.

For more support of time formatting, it would be nice to include custom formatting with [strftime](https://linux.die.net/man/3/strftime) variables. This would make it easily customizable for anyone's preference.

giloser, 2021-02-28:
I'll try to take a look if I get the time but you are welcome to create a PR with the change you would like.

giloser, 2021-03-02:
ok I got some times , should look like that in the next release
[image]

with in the settings :
[image]

Quintin9801, 2021-03-02:
This is perfect! Exactly what I was hoping for, Thank you so much!

closed 2021-03-02 by Quintin9801