jacopotediosi/OctoPrint-Telegram #350: "not connected" message with printer that auto connects.

echan42, 2021-03-16:
My printer is set-up to connect automatically after a /on (m80) command.
When I send my printer a command while the printer is not connected I get the followin message:
"Not connected to a printer. Use /con to connect"
While this statement is true for most use case scenarios it does not apply to mine since what I need to connect is to issue a /on command.

Is there a change you can enable "non connected status" to be edited like every other message?
Thanks!

giloser, 2021-03-17:
I'm sorry but not it's not editable and I don't think I'll change this behavior.
Even if your printer autoconnect if for some reason it doesn't i 'll be on and the command to connect to your printer is still /con

echan42, 2021-03-17:
Understandable, thanks for the response.

closed 2021-03-17 by echan42