jacopotediosi/OctoPrint-Telegram #351: Problem uploading .zip files via telegram

wampattack, 2021-03-18:
[image]

Tried sending a 58KB compressed file with the g-code, but it fails. I have been successful uploading (small) g-code files via telegram, but when I try compressing the same files it won't work. 

Is there any fix to this? 

Since the bot has limitations regarding the size of the files, was hoping I could zip and send larger g-codes via telegram.

wampattack, 2021-03-20:
Here is the log:

2021-03-19 20:54:46,196 - octoprint.plugins.telegram.listener - INFO - Got a command: '/upload' with parameter: '' in chat 1673767930
2021-03-19 20:55:12,742 - octoprint.plugins.telegram.listener - INFO - zip
2021-03-19 20:55:19,271 - octoprint.plugins.telegram.listener - INFO - read archive /home/pi/.octoprint/data/telegram/tmpzip/castle_01mm_pla_mk3_13h55m.zip
2021-03-19 20:55:19,274 - octoprint.plugins.telegram.listener - INFO - namelist 
2021-03-19 20:55:19,599 - octoprint.plugins.telegram.listener - ERROR - Exception occured during processing of a file: Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_telegram/__init__.py", line 229, in handleDocumentMessage
    f.write(data)
TypeError: write() argument must be str, not bytes
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_telegram/__init__.py", line 229, in handleDocumentMessage
    f.write(data)
TypeError: write() argument must be str, not bytes
2021-03-19 20:55:19,599 - octoprint.plugins.telegram.listener - INFO - will now close the zip file
2021-03-19 20:55:20,318 - octoprint.plugins.telegram.listener - INFO - Exception occured during processing of a file: NoneType: None

giloser, 2021-03-20:
hey sorry yes I know the problem I will fix it on the next release.
I hope I will get the time to do it next week.

giloser, 2021-03-23:
Please can you check the #352 (v1.6.5) fix your problem?

wampattack, 2021-03-24:
Just uploaded a compressed file to my octoprint via telegram. Thanks a lot, you rock.

closed 2021-03-24 by wampattack