jacopotediosi/OctoPrint-Telegram #353: [ERR GET IMAGE]

thanatas, 2021-03-28:
After I updated to 1.6.5 I tried to get error in messages.

Meanwhile I use multi-cam plugin also. My stream's adresses are relative, I mean;

/webcamd/?stream=action 
/webcamd2/?stream=action  (I tried to use for ngrok - proxy all the pi)
I haven gone over all code yet, how i can fix or we can fix that?

giloser, 2021-04-01:
you have image in the octoprint interface? for the two?

thanatas, 2021-04-02:
I figured out!

at Webcam & Timelapse  section, I must enable timelapse recordings to work out. But I also noticed that, this part randomly resets the values. Maybe it is octoprint issue, I do not find any log or something. I is working well know. 

But I found such error in the logs :  octoprint.plugins.telegram - ERROR - Exception on_startup: 'NoneType' object has no attribute 'port'

szobov, 2021-04-14:
> octoprint.plugins.telegram - ERROR - Exception on_startup: 'NoneType' object has no attribute 'port'

It's known log message, just ignore it

closed 2021-06-04 by giloser