jacopotediosi/OctoPrint-Telegram #354: plugin_pause_for_user_event_notify causes octoprint to get stuck

WoLpH, 2021-04-13:
When I have the telegram plugin enabled it consistently causes octoprint to stop doing anything whenever my MMU is reporting an error (i.e. no filament or whatever).
Normally after loading the filament the print should continue, but with the telegram plugin enabled it just stops responding entirely. The only solution is to restart octoprint which causes a failed print.

The last message I see in the terminal is: `echo:busy: paused for user`

WoLpH, 2021-04-13:
Little addition, I've just tested 1.6.2 and that one still works
I'm doing a binary search for the broken version and will test 1.6.4 now

WoLpH, 2021-04-13:
When the `plugin_pause_for_user_event_notify` notify is disabled, there's no problem either :)

So for now I'm running 1.6.5 with that disabled. If you need help testing a potential fix, let me know. It's easily reproducible.

giloser, 2021-04-19:
so I did change in version 1.6.3 'add notification wait for user directly on the telegram plugin without need of a other plugin'
because until that version this notification was manage by another plugin but the plugin is still in python 2 so I made some change to not need this one anymore.

let's try to find for where comes the problem because I have everything activate on mine and didn't get trouble from that.
do you have action defined in octoprint? so you can resume the print from octoprint?
do you have the M600 notification activate?
could you send me log file with the debug activate and the problem has occur?

thanks

WoLpH, 2021-04-21:
- do you have action defined in octoprint? so you can resume the print from octoprint?

No, because there is no need. Once the MMU problem is resolved it normally automatically resumes. By default octoprint just keeps waiting forever until the printer says it's done and will continue after that.

- do you have the M600 notification activate?

No.

> could you send me log file with the debug activate and the problem has occur?

Yes and no... even though this problem was consistently reproducible before it does not appear to occur right now. When I was trying to find the plugin that caused the issue I have repeated the same test at least 25 times with consistent failures.

I do notice now that during the time it was not working, I didn't get any telegram messages from the bot either.
Could it be that it only gets stuck if the telegram servers are having issues? I haven't had any issues connecting to telegram from the regular client, but I do notice a gap in the telegram-bot messages.

I will attach the currnet logfile of resuming after the `echo:busy: paused for user` anyhow. Perhaps it is still useful as I'm certain that there's still a bug in some scenarios.
[octoprint.log](https://github.com/fabianonline/OctoPrint-Telegram/files/6353634/octoprint.log)

jacopotediosi, 2025-08-22:
I can’t reproduce this on the latest release [1.9.0](https://github.com/jacopotediosi/OctoPrint-Telegram/releases/tag/1.9.0), using a Prusa MK3S+ with an MMU3.

When the MMU detects a filament runout, I receive the following notifications in this order:
- PrintPaused
- plugin_pause_for_user_event_notify

When I reinsert the filament:
- PrintResumed

Closing since I can’t reproduce the issue and the user did not provide useful logs.

closed 2025-08-22 by jacopotediosi