jacopotediosi/OctoPrint-Telegram #356: listed commands are invalid

sarptekbas, 2021-04-26:
When I try to change the commands of my bot, the Botfather says: 
```Sorry, the list of commands is invalid. Please use this format:

command1 - Description
command2 - Another description

You can also send /empty to clear the list or keep it empty.```
It goes awaw when I delete ```gcode_XXX - Send the gcode to the printer where xxx is the gcode command (/gcode_M117 HelloWorld for exemple)```.
How can I use that command and not get that error?

giloser, 2021-04-26:
Hi,

yes I'll change the documentation. You just have to remove the XXX the command doesn't accept the uppercase

sarptekbas, 2021-04-26:
Oh, ok. So if I only remove the XXX it will work normally? And I can send GCODE Commands?

closed 2021-04-26 by sarptekbas