jacopotediosi/OctoPrint-Telegram #357: Octoprint webserver unable to shutdown with plugin enabled.

phpsystems, 2021-04-30:
Hi,

I recently updated to Octoprint Version 1.6.0 on my raspberry pi. As soon as I did, Octoprint was unable to shutdown cleanly. On investigation, I found that the process was still running, and had to be forced (ie, killed) to shut down cleanly.

The service restarted cleanly (several times) in safe mode, so not an issue with Octoprint. Disabling the telegram plugin allowed for a clean restart.

Details from the logs:

2021-04-28 14:54:59,029 - octoprint.plugins.softwareupdate - INFO - Saved version cache to
disk
2021-04-28 14:54:59,032 - octoprint.plugins.softwareupdate - INFO - Restarting...
2021-04-28 14:54:59,159 - octoprint.server - INFO - Shutting down...
2021-04-28 14:55:00,080 - octoprint.server - INFO - Calling on_shutdown on plugins
2021-04-28 14:55:00,083 - octoprint.events - INFO - Processing shutdown event, this will be
 our last event
2021-04-28 14:55:00,134 - octoprint.events - INFO - Event loop shut down
2021-04-28 14:55:04,708 - octoprint.plugins.telegram - ERROR - An Exception in get layers from DisplayLayerProgress : HTTPConnectionPool(host='localhost', port=5000): Read timed out. (read timeout=3)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 1758, in get_current_layers
    r = requests.get("http://localhost:"+ str(self.tcmd.port) +"/plugin/DisplayLayerProgress/values", headers=headers,timeout=3, proxies=self.getProxies())
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 76, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 530, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/raven/breadcrumbs.py", line 341, in send
    resp = real_send(self, request, *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 643, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/adapters.py", line 529, in send
    raise ReadTimeout(e, request=request)
ReadTimeout: HTTPConnectionPool(host='localhost', port=5000): Read timed out. (read timeout=3)

giloser, 2021-04-30:
I just update myself and try it without problem ...
is this the end of your log? could you activate the debug and try again to see if we get more info ?

thanks

giloser, 2021-06-03:
thanks for the log.
Could you try with the shutdown message disactivate to be sure the problem is there?
I'll add an options to disactivate the check on DisplayLayer but I have tried to disactivate it but it didn't reproduce your problem.

thanks

jacopotediosi, 2025-08-17:
This bug likely shares the same root cause as #442, a bug in OctoPi. However, starting from the upcoming version 1.9.0, template variables in notification messages will be lazy-loaded (b4655665e01e4f73dcdc8e0c8eaac0dc56b16d31). As a result, the DisplayLayerProgress plugin APIs will only be called if a shutdown notification actually contains `{currentLayer}` or `{totalLayer}` placeholders.

closed 2025-08-17 by jacopotediosi