jacopotediosi/OctoPrint-Telegram #363: PSU plugin control not turning on/off printer if octorprint is not on default web port.

cruzer619, 2021-06-11:
not sure if this is an error or feature request.  But code is not grabbing what my server is set to.  IE.. 192.168.1.10:5001  so command is not turning on my printer.  If I manually do the api request via ssh terminal window works fine.

giloser, 2021-10-05:
please try to install that version and test
https://github.com/giloser/OctoPrint-Telegram/archive/refs/heads/master.zip

GonzaloDutra, 2021-10-14:
Hello how are you, how did you manage to vicnulate a GPIO using PSU and also telegram, I have configured my Reley in GPIO2 and it works with PSU, but what command do I have to add or enter so that telegram recognizes the power on of my printer. Thank you

closed 2025-07-11 by jacopotediosi