jacopotediosi/OctoPrint-Telegram #365: Plugin missing after reboot

KochC, 2021-06-22:
When I install the plugin, I reboot the system and can setup everything. After another reboot the log below is shown and the plugin is not shown anymore in the GUI. After another install and reboot everything works again. After another reboot, plugin is missing again and so on.
```
2021-06-22 12:28:15,695 - octoprint.plugins.telegram - ERROR - Exception on_startup: 'NoneType' object has no attribute 'port'
```

NightHawk32, 2021-07-22:
@KochC I see the same behaviour on my setup.

giloser, 2021-08-11:
Hi there,

sorry for the delay.

I see where the problem could be but not why...

I'll add a test and create a test version that I'll post here as soon as I can.
I hope you'll be able to test it.

jacopotediosi, 2025-06-24:
It’s been a few years since this issue was opened. I assume the problem no longer occurs in the [latest version 1.7.0](https://github.com/jacopotediosi/OctoPrint-Telegram/releases/tag/1.7.0), so I’m closing this issue.

If you experience any other bugs, please open a new issue and attach your logs, which you can find in *OctoPrint Settings → Logging → plugin_telegram.log*.