jacopotediosi/OctoPrint-Telegram #372: Add support for Octoprint-Tasmota

augustozanellato, 2021-07-15:
Would be nice to support [this plugin](https://plugins.octoprint.org/plugins/tasmota/), I prefer using this one instead of PSU-Control or Tasmota-MQTT because it also supports power monitoring

giloser, 2021-10-04:
Hi!

I don't have such a plug but I tried to add the support could you test?
https://github.com/giloser/OctoPrint-Telegram/archive/refs/heads/master.zip

augustozanellato, 2021-10-06:
Hi!
That seems to be working fine, only one small note: in Telegram I get a button for each detected plug but the button's text is something like `ipAddressOfThePlug_plugIndex` (example `1.2.3.4_1`) instead of having the name defined in the Tasmota plugin. Not sure if it's an issue with the Tasmota plugin tho

giloser, 2021-10-07:
ok thanks ,
I'll try to take the time to take another look to fix that

commit 6ba6bef0 (2025-08-07): Add support for Tasmota plugin

Close #372
closed 2025-08-20 by jacopotediosi