jacopotediosi/OctoPrint-Telegram #373: Plugin not working after latest update. Bunch of weird messages in the logs.

pjolayres, 2021-07-20:
Recently updated Octoprint to 1.6.1 and Telegram Plugin to 1.6.5. For some reason, the plugin is not working anymore. it neither accepts commands nor reports status to Telegram. I cleared data and reinstalled the plugin with a new Telegram bot and token. Telegram recognizes the token and tests valid (even if logs show a different token). However, it still neither accepts commands nor reports any status. When I open the logs, I get a bunch of weird logs from the plugin:
```
2021-07-20 21:22:45,843 - octoprint.plugins.telegram - ERROR - Exception on_startup: 'NoneType' object has no attribute 'port'
...
2021-07-20 21:30:59,335 - octoprint.plugins.logging - INFO - Setting logger octoprint.plugins.telegram level to 10
2021-07-20 21:22:45,843 - octoprint.plugins.telegram - ERROR - Exception on_startup: 'NoneType' object has no attribute 'port'
2021-07-20 21:32:55,573 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 1...
2021-07-20 21:33:46,220 - octoprint.plugins.telegram - DEBUG - Testing token 1111111111:AAA1_AAAAa-AaaAAAAaaAAAaAAaAaA-1aAa
2021-07-20 21:33:47,435 - octoprint.plugins.telegram - DEBUG - Testing token 1111111111:AAA1_AAAAa-AaaAAAAaaAAAaAAaAaA-1aAa
2021-07-20 21:34:00,024 - octoprint.plugins.telegram - DEBUG - Testing token 1111111111:AAA1_AAAAa-AaaAAAAaaAAAaAAaAaA-1aAa
2021-07-20 21:34:33,896 - octoprint.plugins.telegram - DEBUG - Saving data: {u'message_at_print_done_delay': u'15'}
...
2021-07-20 21:44:54,069 - octoprint.plugins.telegram - ERROR - An Exception in get layers from DisplayLayerProgress : HTTPConnectionPool(host='localhost', port=5000): Read timed out. (read timeout=3)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 2614, in get_current_layers
    proxies=self.getProxies(),
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 76, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/adapters.py", line 529, in send
    raise ReadTimeout(e, request=request)
ReadTimeout: HTTPConnectionPool(host='localhost', port=5000): Read timed out. (read timeout=3)
```

giloser, 2021-10-04:
Hi  do you use the default port for octoprint?
Could you try this version and tell me if it's the same?
https://github.com/giloser/OctoPrint-Telegram/archive/refs/heads/master.zip

mzielinskim, 2021-12-06:
As i had the same messages in my logs i installed your Version and with that the messages are gone.

I use octoPi at Port 80

jacopotediosi, 2025-07-20:
This bug should already be fixed starting from [version 1.7.0](https://github.com/jacopotediosi/OctoPrint-Telegram/releases/tag/1.7.0), in which the method for retrieving the port on which OctoPrint is exposed has been changed.  
If it happens again with the latest version, please open a new issue and follow the instructions in the template, which will guide you in attaching all the necessary information.

closed 2025-07-20 by jacopotediosi