jacopotediosi/OctoPrint-Telegram #377: The bot's /setcommand feature no longer likes the gcode_XXX command

asoretmadolell, 2021-09-01:
I'm trying to copy and paste the list of commands in the readme, and the bot tells me it's not in the correct format. I've narrowed it down to the last command, because if you remove it, it does succeed. I mean the "gcode_XXX" command. I would guess that Telegram has changed how bots work and they don't like underscores or non alphanumeric characters (I tried with parentheses and same thing).

Maybe the command doesn't work and needs to be changed?

ase1590, 2022-05-10:
Seconding this, as its still an issue as of 5/9/22.

The install docs need to be updated.

MarcosKDC, 2023-03-10:
Good afternoon, 

I'm trying this for first time in 2023... 
I'm getting the same problem mentioned here...
Is there already any way to be able to send gcode commands anyway??

Thanks a lot for the time

jacopotediosi, 2025-07-19:
I'm closing this issue because starting from [version 1.8.0](https://github.com/jacopotediosi/OctoPrint-Telegram/releases/tag/1.8.0), setting the bot command list is no longer the user's responsibility — it is now done automatically at each bot startup.  
The section in the README that previously mentioned running `/setcommand` has also been removed.