jacopotediosi/OctoPrint-Telegram #378: Multicam support not working

dirkvranckaert, 2021-09-25:
I have properly setup the multicam support (and that works well in my sidebar etc) and has worked once in Telegram but now it does not anymore (for months already).

Is there anything I can do/review so we can get it working again? Or is it simply broken in the plugin?

chatainsim, 2021-09-28:
Hello, I have the same issue.

giloser, 2021-09-29:
do you have full url define for the webcam link?

chatainsim, 2021-09-29:
Yes for me.
I've removed the multicam plugin for now. It's seems buggy.
Octoprint crashed and I think it's related to multicam plugin.
I will debug later.

Regarding Telegram plugin, is it possible to log in a different log file than octoprint.log ?

Thanks

chatainsim, 2021-09-29:
Here are some logs:
``` 
2021-09-29 15:45:12,446 - octoprint.plugins.telegram.TMSG - DEBUG - Sending Notification: Not printing.
2021-09-29 15:45:12,446 - octoprint.plugins.telegram - DEBUG - start _send_msg
2021-09-29 15:45:12,447 - octoprint.plugins.telegram - DEBUG - PreImgMethod None
2021-09-29 15:45:12,447 - octoprint.plugins.telegram - DEBUG - log instead log sending message 
2021-09-29 15:45:12,448 - octoprint.plugins.telegram - DEBUG - data so far: {'disable_web_page_preview': True, 'chat_id': '372664005', 'disable_notification': False}
2021-09-29 15:45:12,448 - octoprint.plugins.telegram - DEBUG - Snapshot URL: http://admin:admin@192.168.1.44/tmpfs/auto.jpg
2021-09-29 15:45:12,591 - octoprint.plugins.telegram - DEBUG - Image transformations [H:False, V:False, R:False]
2021-09-29 15:45:12,593 - octoprint.plugins.telegram - DEBUG - message so far: Not printing.
2021-09-29 15:45:12,594 - octoprint.plugins.telegram - DEBUG - multicam_profiles : [{'URL': 'http://192.168.1.11:1111', 'flipH': False, 'flipV': False, 'isButtonEnabled': False, 'name': 'Default', 'rotate90': False, 'snapshot': 'http://admin:admin@192.168.1.44/tmpfs/auto.jpg', 'streamRatio': '11:1'}, {'URL': 'http://192.168.1.11:1111/', 'flipH': False, 'flipV': False, 'isButtonEnabled': True, 'name': 'Webcam 1', 'rotate90': False, 'snapshot': 'http://', 'streamRatio': '11:1'}]
2021-09-29 15:45:12,594 - octoprint.plugins.telegram - DEBUG - multicam profile:  {'URL': 'http://192.168.1.11:1111', 'flipH': False, 'flipV': False, 'isButtonEnabled': False, 'name': 'Default', 'rotate90': False, 'snapshot': 'http://admin:admin@192.168.1.44/tmpfs/auto.jpg', 'streamRatio': '11:1'}
2021-09-29 15:45:12,594 - octoprint.plugins.telegram - DEBUG - multicam url :  http://192.168.1.11:1111
2021-09-29 15:45:12,595 - octoprint.plugins.telegram - DEBUG - 192.168.1.11:1111 found so should be replaced by auto.jpg
2021-09-29 15:45:12,595 - octoprint.plugins.telegram - DEBUG - Snapshot URL: http://auto.jpg
2021-09-29 15:45:12,595 - octoprint.plugins.telegram - DEBUG - Snapshot URL: http://auto.jpg
2021-09-29 15:45:12,658 - octoprint.plugins.telegram - ERROR - TimeOut Exception: HTTPConnectionPool(host='auto.jpg', port=80): Max retries exceeded with url: / (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fbe353a1be0>: Failed to establish a new connection: [Errno -2] Name or service not known'))
2021-09-29 15:45:12,658 - octoprint.plugins.telegram - DEBUG - Image for  Default
2021-09-29 15:45:12,804 - octoprint.plugins.telegram - DEBUG - multicam profile:  {'URL': 'http://192.168.1.11:1111/', 'flipH': False, 'flipV': False, 'isButtonEnabled': True, 'name': 'Webcam 1', 'rotate90': False, 'snapshot': 'http://', 'streamRatio': '11:1'}
2021-09-29 15:45:12,805 - octoprint.plugins.telegram - DEBUG - multicam url :  http://192.168.1.11:1111/
2021-09-29 15:45:12,805 - octoprint.plugins.telegram - DEBUG - Snapshot URL: http://192.168.1.11:1111/
2021-09-29 15:45:12,806 - octoprint.plugins.telegram - DEBUG - Snapshot URL: http://192.168.1.11:1111/
2021-09-29 15:48:14,190 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAAa1-a1AA1aaAAAaAaaAaAA1AaAAaAAaAa
2021-09-29 15:48:20,579 - octoprint.plugins.telegram - DEBUG - Saving data: {'multicam': False}
2021-09-29 15:48:23,387 - octoprint.plugins.telegram - DEBUG - self.tmsg = <octoprint_telegram.telegramNotifications.TMSG object at 0x7fbe0827d780>
2021-09-29 15:48:25,187 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAAa1-a1AA1aaAAAaAaaAaAA1AaAAaAAaAa
2021-09-29 15:48:32,899 - octoprint.plugins.telegram - DEBUG - Saving data: {'multicam': True}
2021-09-29 15:48:35,659 - octoprint.plugins.telegram - DEBUG - self.tmsg = <octoprint_telegram.telegramNotifications.TMSG object at 0x7fbe0827d780>
```
I don't know what is 192.168.1.11:1111 and the token is wrong maybe normal for security reason.
Here is my multicam conf:
[image]

giloser, 2021-09-29:
ok thanks.

can you also show how is defined the default webcam in octoprint?

thanks

chatainsim, 2021-09-29:
Here is the default webcam:
[image]

giloser, 2021-10-01:
well I think that it's because there is nothing after the port.
Like we still don't have the porposal from fabian on the multicam to get the a snapshot url, we still try to create it and in your case it's not going very well.
I'll try to made some change and will send you a link for a test version.

chatainsim, 2021-10-03:
Is it possible to use ffmpeg to make a snapshot from a stream ?
Because I have a snapshot URL only on one camera.
Thanks

giloser, 2021-10-04:
well I'm not sure about that.
you will have to check that and also I'll adapt the code to get the snapshot from the option of the multicam plugin.
The problem is we still have not access to this options on the graphic options of multicam.
So you are gonna have to change if in the config file by hand.

config.yaml¶
If not specified via the command line, the main configuration file config.yaml for OctoPrint is expected in its settings folder, which unless defined differently via the command line is located at ~/.octoprint on Linux, at %APPDATA%/OctoPrint on Windows and at ~/Library/Application Support/OctoPrint on macOS. If the file is not there, you can just create it - it will only get created by OctoPrint once you save settings that deviate from the default settings.

giloser, 2021-10-04:
please try to install that version and test
https://github.com/giloser/OctoPrint-Telegram/archive/refs/heads/master.zip

chatainsim, 2021-10-04:
I still have the same issue with this version.
Here are the logs:
```
2021-10-04 22:44:30,123 - octoprint.plugins.telegram - ERROR - Exception on_startup: 'NewTelegramPlugin' object has no attribute 'main'
2021-10-04 22:44:33,023 - octoprint.plugins.telegram - DEBUG - Starting listener.
2021-10-04 22:44:33,029 - octoprint.plugins.telegram.listener - DEBUG - Try first connect.
2021-10-04 22:44:33,180 - octoprint.plugins.telegram - DEBUG - getMe returned: {'ok': True, 'result': {'id': xxxxxxxxx, 'is_bot': True, 'first_name': 'PrinterBot', 'username': 'BotName', 'can_join_groups': True, 'can_read_all_group_messages': False, 'supports_inline_queries': False}}
2021-10-04 22:44:33,180 - octoprint.plugins.telegram - DEBUG - getMe status code: 200
2021-10-04 22:44:33,180 - octoprint.plugins.telegram.listener - DEBUG - Setting status: Connected as @BotName.
2021-10-04 22:44:33,180 - octoprint.plugins.telegram.listener - DEBUG - Listener is running.
2021-10-04 22:44:33,180 - octoprint.plugins.telegram.listener - DEBUG - listener: sending request with offset 0...
2021-10-04 22:44:33,237 - octoprint.plugins.telegram - DEBUG - Testing Pic ID xxxxxxxxx
2021-10-04 22:44:33,238 - octoprint.plugins.telegram - DEBUG - Requesting Profile Photo for chat_id: xxxxxxxxx
2021-10-04 22:44:33,238 - octoprint.plugins.telegram - DEBUG - requests.get(https://api.telegram.org/bot111111111:AAAa1-a1AA1aaAAAaAaaAaAA1AaAAaAAaAa/getUserProfilePhotos
2021-10-04 22:44:33,318 - octoprint.plugins.telegram.listener - DEBUG - Updating update_offset from 0 to 967404590
2021-10-04 22:44:33,373 - octoprint.plugins.telegram - DEBUG - Requesting file with id AgACAgQAAxUAAWFbRa0GydXtCct52jZeNibNssmEAAKqpzEbxWY2Fm8p3VNl0iS-AQADAgADYQADIQQ.
2021-10-04 22:44:33,453 - octoprint.plugins.telegram.listener - DEBUG - Updating update_offset from 967404590 to 967404591
2021-10-04 22:44:33,573 - octoprint.plugins.telegram.listener - DEBUG - Ignoring message because first_contact is True.
2021-10-04 22:44:33,574 - octoprint.plugins.telegram - DEBUG - self.tmsg = <octoprint_telegram.telegramNotifications.TMSG object at 0x7ffab03226a0>
2021-10-04 22:44:33,574 - octoprint.plugins.telegram - DEBUG - Got an event: PrinterStart Payload: {}
2021-10-04 22:44:33,575 - octoprint.plugins.telegram.TMSG - DEBUG - event :PrinterStart
2021-10-04 22:44:33,577 - octoprint.plugins.telegram.TMSG - DEBUG - send_gif = True and this message would send gif = False
2021-10-04 22:44:33,578 - octoprint.plugins.telegram.TMSG - DEBUG - Printer Status{'state': {'text': 'Operational', 'flags': {'operational': True, 'printing': False, 'cancelling': False, 'pausing': False, 'resuming': False, 'finishing': False, 'closedOrError': False, 'error': False, 'paused': False, 'ready': True, 'sdReady': False}, 'error': ''}, 'job': {'file': {'name': None, 'path': None, 'size': None, 'origin': None, 'date': None}, 'estimatedPrintTime': None, 'lastPrintTime': None, 'filament': {'length': None, 'volume': None}, 'user': None}, 'currentZ': None, 'progress': {'completion': None, 'filepos': None, 'printTime': None, 'printTimeLeft': None, 'printTimeOrigin': None}, 'offsets': {'tool0': 0}, 'resends': {'count': 0, 'transmitted': 3, 'ratio': 0}}
2021-10-04 22:44:33,578 - octoprint.plugins.telegram.TMSG - DEBUG - TEMPS - {}
2021-10-04 22:44:33,578 - octoprint.plugins.telegram - DEBUG - get_current_layers api key=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
2021-10-04 22:44:33,597 - octoprint.plugins.telegram - DEBUG - get_current_layers : r=<Response [200]>
2021-10-04 22:44:33,597 - octoprint.plugins.telegram.TMSG - DEBUG - Layers - {'currentFilename': '', 'fanSpeed': '-', 'feedrate': '-', 'feedrateG0': '-', 'feedrateG1': '-', 'height': {'current': '-', 'currentFormatted': '-', 'total': '-', 'totalFormatted': '-'}, 'layer': {'averageLayerDuration': '-', 'averageLayerDurationInSeconds': '-', 'current': '-', 'lastLayerDuration': '-', 'lastLayerDurationInSeconds': '-', 'total': '-'}, 'print': {'changeFilamentCount': 0, 'changeFilamentTimeLeft': '-', 'changeFilamentTimeLeftInSeconds': 0, 'estimatedChangedFilamentTime': '-', 'estimatedEndTime': '-', 'm73progress': '', 'printerState': 'operational', 'progress': '0', 'timeLeft': '-', 'timeLeftInSeconds': '-'}}
2021-10-04 22:44:33,598 - octoprint.plugins.telegram.TMSG - DEBUG - VARS - {'self': <octoprint_telegram.telegramNotifications.TMSG object at 0x7ffab03226a0>, 'payload': {}, 'kwargs': {'event': 'PrinterStart', 'with_image': False, 'with_gif': False, 'silent': False, 'thumbnail': None}, 'status': {'state': {'text': 'Operational', 'flags': {'operational': True, 'printing': False, 'cancelling': False, 'pausing': False, 'resuming': False, 'finishing': False, 'closedOrError': False, 'error': False, 'paused': False, 'ready': True, 'sdReady': False}, 'error': ''}, 'job': {'file': {'name': None, 'path': None, 'size': None, 'origin': None, 'date': None}, 'estimatedPrintTime': None, 'lastPrintTime': None, 'filament': {'length': None, 'volume': None}, 'user': None}, 'currentZ': None, 'progress': {'completion': None, 'filepos': None, 'printTime': None, 'printTimeLeft': None, 'printTimeOrigin': None}, 'offsets': {'tool0': 0}, 'resends': {'count': 0, 'transmitted': 3, 'ratio': 0}}, 'event': 'PrinterStart', 'z': 0.0, 'temps': {}, 'bed_temp': 0.0, 'bed_target': 0.0, 'e1_temp': 0.0, 'e1_target': 0.0, 'e2_temp': 0.0, 'e2_target': 0.0, 'percent': 0, 'Layers': {'currentFilename': '', 'fanSpeed': '-', 'feedrate': '-', 'feedrateG0': '-', 'feedrateG1': '-', 'height': {'current': '-', 'currentFo [... cut ...]
2021-10-04 22:56:45,188 - octoprint.plugins.telegram - DEBUG - multicam url :  http://192.168.1.11:1111/
2021-10-04 22:56:45,188 - octoprint.plugins.telegram - DEBUG - Snapshot URL: http://192.168.1.11:1111/
2021-10-04 22:56:45,188 - octoprint.plugins.telegram - DEBUG - Snapshot URL: http://192.168.1.11:1111/
2021-10-04 22:56:59,153 - octoprint.plugins.telegram - DEBUG - Testing token 111111111:AAAa1-a1AA1aaAAAaAaaAaAA1AaAAaAAaAa
2021-10-04 22:57:07,595 - octoprint.plugins.telegram - DEBUG - Saving data: {'debug': False, 'multicam': False}

```

giloser, 2021-10-04:
did you also change the snapshot url for the second camera and set it in the config.yaml?

chatainsim, 2021-10-04:
I have not.
Because I don't have a snapshot url for the second webcam.

dirkvranckaert, 2021-10-05:
If it can be of any help here are my OctoPrint settings for the built-in camera settings and for the multicam settings.

[image]
[image]

giloser, 2021-10-05:
when you test the snapshot in the timelapse is it working? why is the this a local ip and all the other not? 
are they something you can reach from the outside?
can you give me a log file? did you try the test version I provided?
thanks

dirkvranckaert, 2021-10-05:
No I need to fine a moment when not printing to test this new version.

My OctoPrint is configured with ngrok to be reachable from the outside indeed, for the stream url and the multicam I needed to provide the public accessible URL in order for my mobile apps to be able to give me a stream of my prints. The timelapse recording url works (however I disabled it since I don't use it actually but I did test is a week ago). Don't know why that needs to be the local address, but if I provide there my ngrok address it does not work when pressing the test button.

I will give a try to the update you provided hopefully somewhere this evening

giloser, 2021-10-05:
well I didn't do the snapshot part but as far as I understand we work like the timelapse part so if it doesn't work for the timelapse it will probably not also in the telegram snapshot.
Maybe by adding a snapshot url in the config.yaml for the multicam plugin (I'll try to get an update to have access to the snaphsot url in the plugin configuration) that is local it could work.

dirkvranckaert, 2021-10-05:
Ah ok, I could indeed set local IP's, let me know when you have an update ready, I'll be happy to test :-)

giloser, 2021-10-05:
well I send a pull request on the multicam plugin.
if you want to try you could already change it by hand in the config.yaml and update to the test version I send earlier.

dirkvranckaert, 2021-10-05:
@giloser This test version of the multicam plugin contains your PR changes already, right? So I don't have to manually edit something?

giloser, 2021-10-05:
yes this version should give you access on those option in the graphical options

jacopotediosi, 2025-06-26:
The [1.7.0 release](https://github.com/jacopotediosi/OctoPrint-Telegram/releases/tag/1.7.0) includes a complete rework of webcam handling and the implementation of OctoPrint's new webcam integration. This should have resolved all issues related to webcams, multicam setups, and webcam orientation.

If you still encounter any bugs, please open a new issue and attach the logs found in *OctoPrint Settings → Logging → plugin_telegram.log*.

Thank you for your report.