jacopotediosi/OctoPrint-Telegram #379: Got KeyError after adding new command.

unixneo, 2021-10-04:
Read the docs (maybe I missed something).

Added the following in telegramCommands.py:

```python
 self.commandDict = {
            ....,
            "/stat": {"cmd": self.cmdStat},  # unixneo 4/10/21  add start command (status without image)
            ....,
}

self.main.send_msg(
            self.gEmo("info")
            + gettext(
               ....,
                "/stat - Sends the current status with no photo.\n"
              ....,
            )
)

  ############################################################################################
    def cmdStat(self, chat_id, from_id, cmd, parameter, user = ""):
        if not self.main._printer.is_operational():
            self.main.send_msg(
                self.gEmo("warning")
                + gettext(" Not connected to a printer. Use /con to connect."),
                chatID=chat_id,
                inline=False,
            )
        elif self.main._printer.is_printing():
            self.main.on_event("StatusPrinting", {}, chatID=chat_id)
        else:
            self.main.on_event("StatusNotPrinting", {}, chatID=chat_id)
```

But I keep getting this error:

```shell
  File "/Applications/OctoPrint/venv/lib/python3.7/site-packages/octoprint_telegram/__init__.py", line 2539, in isCommandAllowed
    if self.chats[chat_id]["commands"][command]:
KeyError: '/stat'
```

Is there somewhere else I need to bind this new command?

giloser, 2021-10-04:
The new command should be listed in the command list on the web interface of octoprint telegram plugin page.

you have to enable your new command.

unixneo, 2021-10-04:
Thanks.  I edited the permissions of the user and checked the new command and it is not giving an error now.

I wonder why, however, this code is still sending a snapshot for the new command (which is basically the status command without the image)

```python
 def cmdStat(self, chat_id, from_id, cmd, parameter, user = ""):
        if not self.main._printer.is_operational():
            self.main.send_msg(
                self.gEmo("warning")
                + gettext(" Not connected to a printer. Use /con to connect."),
                chatID=chat_id,
                inline=False,
            )
        elif self.main._printer.is_printing():
            self.main.on_event("StatusPrinting", {}, chatID=chat_id)
        else:
            self.main.on_event("StatusNotPrinting", {}, chatID=chat_id)
```

[image]
I

giloser, 2021-10-04:
you use the event StatusPrinting without forcing options so he will select the options you select in the graphic interface and if the image are activate he will take an image.

you'll need to send something like

with_image=with_image

unixneo, 2021-10-04:
Thanks again.  I better go back and read more docs.

closed 2021-10-04 by unixneo

unixneo, 2021-10-04:
Got it working now, a new /stat command without an image ... thanks again @giloser 

[image]

unixneo, 2021-10-04:
Just in case anyone is interested in how I coded this:

https://github.com/unixneo/OctoPrint-Telegram/commit/c0010a24a6dccbc81e8ed5f99e228abed319032a