jacopotediosi/OctoPrint-Telegram #384: Plugin messes up my camera stream

Natelicious, 2021-11-08:
**Initially before a print**

- Camera stream works in Octoprint, I can see the video in the control tab.
- In the telegram bot, everything works well. I can see the image using /status.

**After initiating a print using Octoprint**

- Telegram cant take a picture and I get "ERR GET IMAGE"
- My video stream in Octoprint Control tab states "Webcam stream loading..." forever.
- In the settings, "/webcam/?action=stream doesn't display" anything
- In the settings, "http://127.0.0.1:8080/?action=snapshot doesn't display" anything

Octoprint version: 1.7.2
Octopi: 0.18
Camera: Raspi Cam (Did not edit any settings in putty)

Is there any solution out there? I 've reflashed OctoPi and tried again but the same result still happened. Thank you!

giloser, 2021-11-10:
Hi

did you tried a print with telegram disabled and that create no problem?
you said the reboot didn't fix the problem? so how do you fix the problem?

could you send a logfile of a print with telegram plugin activate?

thanks

rdeanar, 2021-11-16:
I had the same problem after upgrading plugin to the version with implemented socks5 proxy. 

Some time ago before upgrade i manually merge [this PR with proxy support](https://github.com/fabianonline/OctoPrint-Telegram/pull/128) to local version of plugin, so proxy urls was already configured  in my config.

After upgrade bot stopped to send images from the webcam with `[ERR GET IMAGE]` error, however camera and snapshots in octoprint works fine.

I tried to figure out the problem, enabled debug logs and saw:
```
2021-11-16 09:38:58,115 - octoprint.plugins.telegram.listener - INFO - Got a command: '/status' with parameter: '' in chat 12345678
2021-11-16 09:38:58,126 - octoprint.plugins.telegram - DEBUG - start _send_msg
2021-11-16 09:38:58,132 - octoprint.plugins.telegram - DEBUG - PreImgMethod None
2021-11-16 09:38:58,136 - octoprint.plugins.telegram - DEBUG - log instead log sending message
2021-11-16 09:38:58,139 - octoprint.plugins.telegram - DEBUG - data so far: {u'disable_notification': False, u'chat_id': '12345678', u'disable_web_page_preview': True}
2021-11-16 09:38:58,143 - octoprint.plugins.telegram - DEBUG - Snapshot URL: http://127.0.0.1:1111/?action=snapshot
2021-11-16 09:38:58,680 - octoprint.plugins.telegram - ERROR - TimeOut Exception: SOCKSHTTPConnectionPool(host='127.0.0.1', port=8080): Max retries exceeded with url: /?action=snapshot (Caused by NewConnectionError('<urllib3.contrib.socks.SOCKSConnection object at 0x65679bf0>: Failed to establish a new connection: 0x05: Connection refused',))
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 1722, in take_image
    r = requests.get(snapshot_url,timeout=10, proxies=self.getProxies())
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
ConnectionError: SOCKSHTTPConnectionPool(host='127.0.0.1', port=8080): Max retries exceeded with url: /?action=snapshot (Caused by NewConnectionError('<urllib3.contrib.socks.SOCKSConnection object at 0x65679bf0>: Failed to establish a new connection: 0x05: Connection refused',))
2021-11-16 09:38:58,896 - octoprint.plugins.telegram - DEBUG - image data so far: None
2021-11-16 09:38:58,898 - octoprint.plugins.telegram - DEBUG - message so far: [ERR GET IMAGE]

 ⚠ Not connected to a printer. Use /con to connect.
2021-11-16 09:38:58,900 - octoprint.plugins.telegram - DEBUG - Sending without image.. 249821329
2021-11-16 09:39:00,278 - octoprint.plugins.telegram - DEBUG - Sending finished. 200
```

I had checked my proxy logs:
```
2021/11/16 09:05:14 [ERR] socks: Failed to handle request: Connect to 127.0.0.1:8080 failed: dial tcp 127.0.0.1:8080: getsockopt: connection refused
2021/11/16 09:05:14 [ERR] socks: Failed to handle request: Connect to ::1:5000 failed: dial tcp [::1]:5000: connect: cannot assign requested address
```

It seems that plugin tried to get image through proxy instead of direct connection to localhost.

In conclusion, i have disabled proxy in plugin settings and i was able to see image in bot response message.

Natelicious, 2021-11-17:
> Hi
> 
> did you tried a print with telegram disabled and that create no problem? you said the reboot didn't fix the problem? so how do you fix the problem?
> 
> could you send a logfile of a print with telegram plugin activate?
> 
> thanks

I will try and let you know again. I didnt try with telegram disabled and after rebooting, I still get the camera error.

Natelicious, 2021-11-17:
@giloser I've managed to get it to work!

However I followed the same steps that I took for setting up the plugin the last 3 times. I am unsure of how it worked this time round. Thank you all for your help anyway. Stay safe!

closed 2021-11-17 by Natelicious