jacopotediosi/OctoPrint-Telegram #385: Faulty output of height with the printing end with 10mm

Newbie-2, 2021-11-29:
Whenever the pressure is finished, I'll leave with me component height with {z}.
However, the same height is spent 10mm with each component, which is wrong.
The information of the height {z} during a layer change we output correctly each time.

Specification
-----------------------------------------------------------
PrintDone:
Druck erfolgreich fertig    {emo:check}
{file}  
Druckzeit:  {time_done} 
Bauteil Höhe: {z} mm
-----------------------------------------------------------
ZChange:
-- Z-Change --
Bed-Temperatur:  {bed_temp} °C
Extruder-Temperatur:  {e1_temp} °C
Aktuelle Höhe: {z} mm
Druck-Ende:  {time_finish}
Verbrauchte-Zeit:  {time_done} = {percent}%
Rest-Druckzeit: {time_left}

Newbie-2, 2023-02-06:
Doesn't another user have the same problem?

jacopotediosi, 2025-08-09:
A personal reminder: when we integrate SlicerSettingsParser support, it would be nice to also show the total height of objects in the details files in response to the `/files` command, as we already do, for example, for cost estimation.

closed 2025-09-18 by jacopotediosi